blue-systems/plamo-initial-dev_DEPRECATED #13: Virtual Keyboard: Malitt

star-buck, 2015-06-07:


boudewijnrempt, 2015-06-08:
# Running notes on maliit

# Maliit

* http://maliit.org/ no longer exists, and the wiki is gone, too.

# Enabling Maliit

* Using maliit should be pretty simple, just link plasma-active:

export QT_IM_MODULE=Maliit
maliit-server -software -bypass-wm-hint

(from http://ppenz.blogspot.com/2012/04/plasma-active-and-maliit.html)

Except that the commandline options have changed. 

Maliit + ubuntu-keyboard on Vivid segfaults:

    apt-get install ubuntu-keyboard
    export QT_IM_MODULE=Maliit
    maliit-server 
    CRITICAL: void MaliitKeyboard::Logic::WordEnginePrivate::loadPlugin(QString, QString)  Loading plugin failed:  "The shared library was not found."
    WARNING: virtual void MaliitKeyboard::Logic::WordEngine::setWordPredictionEnabled(bool) No backend available, cannot enable word engine!
    Segmentation fault (core dumped)

Ubuntu Maliit + maliit-keyboard or nemo-keyboard: the ubuntu: maliit keeps looking for /usr/lib/maliit/plugins/libubuntu-keyboard-plugin.so. (Where's the active plugin configured? Or is it hard-coded in Ubuntu?)

# Building Maliit from https://github.com/maliit

https://github.com/maliit/maliit.git does not build. 

With CONFIG+=wayland:

g++ -c -m64 -pipe -O2 -fPIC -Wall -W -D_REENTRANT -DMALIIT_PLUGINS_DIR=\"/usr/lib/maliit/plugins\" -DMALIIT_PLUGINS_DATA_DIR=\"/usr/share/maliit/plugins\" -DMALIIT_FACTORY_PLUGINS_DIR=\"/usr/lib/maliit/plugins/factories\" -DMALIIT_EXTENSIONS_DIR=\"/usr/share/maliit-framework/extensions/\" -DMALIIT_TEST_PLUGINS_DIR=\"/usr/lib/maliit-framework-tests/plugins\" -DMALIIT_CONFIG_ROOT=\"/maliit/\" -DMALIIT_FRAMEWORK_USE_INTERNAL_API -DHAVE_WAYLAND -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_GUI_LIB -DQT_CORE_LIB -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I.moc -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o .obj/dummy.o dummy.cpp
dummy.cpp:1:50: fatal error: wayland-input-method-client-protocol.h: No such file or directory
 #include "wayland-input-method-client-protocol.h"
                                                  ^
compilation terminated.
Makefile:515: recipe for target '.obj/dummy.o' failed
make[1]: *** [.obj/dummy.o] Error 1

With CONFIG+=enable-dbus-activation:

maliit/plugins/updateevent.cpp: In member function ‘bool MImUpdateEventPrivate::isFlagSet(Qt::InputMethodHint, bool*) const’:
maliit/plugins/updateevent.cpp:42:107: error: conversion from ‘qlonglong {aka long long int}’ to ‘QFlag’ is ambiguous
                                              update.value(Maliit::Internal::inputMethodHints).toLongLong()));
                                                                                                           ^
maliit/plugins/updateevent.cpp:42:106: note: candidates are:
                                              update.value(Maliit::Internal::inputMethodHints).toLongLong()));
                                                                                                          ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:1085:0,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qnamespace.h:37,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs.h:41,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:40,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QObject:1,
                 from ./maliit/plugins/extensionevent.h:18,
                 from ./maliit/plugins/updateevent.h:18,
                 from maliit/plugins/updateevent.cpp:15:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:65:29: note: QFlag::QFlag(ushort)
     Q_DECL_CONSTEXPR inline QFlag(ushort ai) : i(int(uint(ai))) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:64:29: note: QFlag::QFlag(short int)
     Q_DECL_CONSTEXPR inline QFlag(short ai) : i(int(ai)) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:63:29: note: QFlag::QFlag(uint)
     Q_DECL_CONSTEXPR inline QFlag(uint ai) : i(int(ai)) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:52:29: note: QFlag::QFlag(int)
     Q_DECL_CONSTEXPR inline QFlag(int ai) : i(ai) {}
[... 14 lines cut ...]
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:40,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QObject:1,
                 from ./maliit/plugins/extensionevent.h:18,
                 from ./maliit/plugins/updateevent.h:18,
                 from maliit/plugins/updateevent.cpp:15:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:65:29: note: QFlag::QFlag(ushort)
     Q_DECL_CONSTEXPR inline QFlag(ushort ai) : i(int(uint(ai))) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:64:29: note: QFlag::QFlag(short int)
     Q_DECL_CONSTEXPR inline QFlag(short ai) : i(int(ai)) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:63:29: note: QFlag::QFlag(uint)
     Q_DECL_CONSTEXPR inline QFlag(uint ai) : i(int(ai)) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:52:29: note: QFlag::QFlag(int)
     Q_DECL_CONSTEXPR inline QFlag(int ai) : i(ai) {}
                             ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:113:29: note: initializing argument 1 of ‘QFlags<T>::QFlags(QFlag) [with Enum = Qt::InputMethodHint]’
     Q_DECL_CONSTEXPR inline QFlags(QFlag f) : i(f) {}
                             ^
maliit/plugins/updateevent.cpp:94:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

# Building Maliit from https://launchpad.net/maliit

boudewijnrempt, 2015-06-08:
## Ubuntu

bzr branch lp:ubuntu/vivid/maliit-plugins
bzr branch lp:ubuntu/vivid/maliit-frameworks

### examples/apps/plainqt

Need to remove the obsolete "-bypass-wm-hint" argument. On starting the maliit server, the plugin gets loaded correctly, but no keyboard pops up...

## Github upstream

https://github.com/maliit/framework
https://github.com/maliit/plugins

The plugins don't configure because /usr/share/qt5/mkspecs/features/maliit-defines.prf doesn't get loaded. After adding the contents to the config.pri file, the maliit-keyboard cpp-based plugin doesn't build because the framework includes are in the wrong place; after fixing that, the cpp-based plugin doesn't link because of undefined references like

../common/libtests-common.a(moc_inputmethodhostprobe.o):(.data.rel.ro._ZTI20InputMethodHostProbe[_ZTI20InputMethodHostProbe]+0x10): undefined reference to `typeinfo for MAbstractInputMethodHost'

### examples/apps/plainqt

Has been updated to already remove the obsolete argument.

On starting the server from the plaintqt example, the plugin gets loaded, but nothing happens.

boudewijnrempt, 2015-06-09:
The keyboard startup problem was caused by not finding the plugin. This change fixed the issue:

 class MaliitPlatformInputContextPlugin: public QPlatformInputContextPlugin
 {
     Q_OBJECT
-    Q_PLUGIN_METADATA(IID "org.qt-project.Qt.QPlatformInputContextFactoryInterface" FILE "maliit.json")
+    Q_PLUGIN_METADATA(IID QPlatformInputContextFactoryInterface_iid FILE "maliit.json")

We are using our own forks for malit-frameworks now.

boudewijnrempt, 2015-06-11:
Current state:

* patched QtWayland that activates the right plugin
* maliit-frameworks and maliit-plugins taken from Marco's fork from the plasma-active days
* when testing on desktop, keyboard shows up both in X11 and wayland
* aftyer settiong up env. it should work on phone

but: lots of kwin work needed:

* kwin needs to know which window is the virtual keyboard to ensure that it's on top of the stack and not below the entry field

Not as kwin problem:

* the screen must be rearranged so the entry field and the keyboard don't overlap --
the keyboard creates a full-screen keyboard that is half-translucent: query, is this maliit, or nemo-keyboard?

boudewijnrempt, 2015-06-12:
Kwin needs to know which window is the keyboard so it always shows it on top and never gives it keyboard focus.

star-buck, 2015-06-13:
update:
* kwin
* kwayland
* plasma-maliit-frameworks
* plasma-maliit-plugins

star-buck, 2015-06-14:
v102: keyboard still doesnt show up.

mgraesslin, 2015-06-14:
problem is that kwinwrapper is missing 
"--inputmethod maliit-server"
in the command to start kwin.

To edit manually: sudo vim /usr/bin/kwinwrapper
and change the kwin start line to:
kwin_wayland --hwcomposer --xwayland --libinput --inputmethod maliit-server plasma-phone 2> /tmp/kwin_logs

star-buck, 2015-06-17:
keyboard shows up as of 104.

closed 2015-06-17 by star-buck