blue-systems/plamo-initial-dev_DEPRECATED #14: Home/Tasks/X Buttons

star-buck, 2015-06-07:


notmart, 2015-06-11:
kwin is crashing with the switcher plugin.
@mgraesslin is investigating on it, it looks like is a libhybris limitation, so a workaround on it is being searched for

mgraesslin, 2015-06-11:
The crash is in fact an abort in QtQuick Scene Graph because QtWayland fails to create an EGLDisplay. The reason for that is a limitation in libhybris. Libhybris loads the actual windowing system egl implementation as a plugin. When KWin starts it creates an EGLDisplay for the windowing system "hwcomposer". QtQuick/QtWayland try to create an EGLDisplay for the windowing system "wayland". As libhybris doesn't support multiple windowing systems in one process, the call goes into the hwcomposer backend's eglGetDisplay which expects no argument passed into it, but it gets a wl_display* and fails. Without the EGLDisplay QtWayland is not able to create an OpenGL context, without an OpenGL context QtQuick aborts.

I see two possible solutions and for both I do not know whether that is possible:
1. remove the limitation from libhybris
2. Create a plugin for QtWayland to create the OpenGL context

For the first the problem is that we don't have any "meta" information about the EGLDisplay to be created. The argument passed into eglGetDisplay is of type EGLNativeDisplayType which is just a void*. So I do not see how one could create logic based on the call to load a different library.

The second has the problem that I still need to do the buffer exchange protocol. And that actually needs the wayland egl platform.

luebking, 2015-06-11:
> So I do not see how one could create logic based on the call to load a different library.
If not already used for other purposes, one could perhaps tag[1] the pointer (from kwin to call into hwcomposer)
This requires research on the actual usage, since "void" data is naturally not somehow aligned.

[1] http://en.wikipedia.org/wiki/Tagged_pointer

mgraesslin, 2015-06-11:
after discussion with @sebasje @notmart and @star-buck we decided to delay the problem and try to make the panel as part of Plasma with a short term solution of sending kwin scripts to close the active window.

notmart, 2015-06-11:
partial alternative solution implemented in plasmashell (close button and home button, window list missing) needs update of plasma-phone-components and kwin.

closed 2015-06-13 by star-buck