blue-systems/plamo-initial-dev_DEPRECATED #9: allow set wallpaper

star-buck, 2015-06-07:


notmart, 2015-08-07:
the wallpaper dialog now has an "add wallpaper" button that shows a dialog with a list of wallpapers from the ~/Pictures folder, it makes at least easier to add custom wallpapers on the device.

It's still not perfect in many ways, besides being maybe desiderable to set it from other applications (quite a difficult problem)
we probably want to make dialogs like that a frameworksintegration plugin that replaces the standard KDE file picker dialog with something that just shows a list of "all the pictures" "all the songs" etc.

star-buck, 2015-09-24:
can we add this also to some kcm, so the wallpaper becomes changeable additionally from Settings?

closed 2015-11-15 by star-buck