blue-systems/plamo-initial-dev_DEPRECATED #21: install packages without root

star-buck, 2015-06-08:


star-buck, 2015-06-10:
Example: 
Ubuntu Touch .click-format: http://ubuntuphone.es/app-store/app/com.ubuntu.scopes.soundcloud

star-buck, 2015-06-16:
Good info about click vs. Deb:
https://lwn.net/Articles/565521/

aleixpol, 2015-06-17:
Installing without root password would be also good from an SDK PoV. We can't specify the password if we call "adb shell sudo dpkg -i mypkg.deb".

star-buck, 2015-06-17:
video with ome more insight about ubuntu click: 
https://www.youtube.com/watch?v=h1anchaDI-E

star-buck, 2015-06-17:
Spanish Click App-Store:
http://ubuntuphone.es/app-store/

aleixpol, 2015-06-17:
To install a click file:
```
pkcon install-local file.click --allow-untrusted
```

Look for desktop files in:
```
#username/appname/appname.desktop
/opt/click.ubuntu.com/.click/users/*/*/*.desktop
/usr/share/click/preinstalled/.click/users/*/*/*.desktop
```

Still need to see how to execute the applications.

aleixpol, 2015-06-17:
Running an application is done by:
```
ubuntu-app-launch messaging-app
```
It works on the ubuntu phone, doesn't work on my Kubuntu mobile image.

aleixpol, 2015-06-18:
Been playing a bit further, my conclusion is that click already works, but we need to be able to integrate applications written in click.

A first step will be to include ```/userdata/user-data/phablet/.local/share/applications/``` in the launcher list. This won't work on all applications though (e.g. the gallery app) because to execute the applications they use ```aa-exec-click``` that checks whether it's running Mir. This will make the desktop files not to work sometimes. Nevertheless, some applications such as the one described below will just work.

For further research, here's how to fetch a random click file:
```
from oauthlib.oauth1.rfc5849 import Client
import requests

# curl https://login.ubuntu.com/api/v2/tokens/oauth -d '{"email": "aleixpol@kde.org", "password": "<password>", "token_name": "click_example"}' -H 'Content-Type: application/json'
#{
    #"openid": "myGdLpz",
    #"token_name": "click_example",
    #"date_updated": "2015-06-18T00:21:09.109",
    #"token_key": "XLWbzoFAgX5Hdg0ViNS9pd55wYdH7THdIOaSVPaphtiONjnKSE02JagonMx6",
    #"consumer_secret": "7CyRE2o10a3ji7Or5QdkRET4mORPHpBcZdgVrv8l1pdEfqRIFv",
    #"href": "/api/v2/tokens/oauth/XLWbzoFAgX5Hdg0ViNS9pd55wYdH7THdIOaSVPaphtiONjnKSE02JagonMx6",
    #"date_created": "2015-06-18T00:21:09.109",
    #"consumer_key": "myGdLpz",
    #"token_secret": "0d5hQOhUuQ3U5oOmqs2Nbmy8lcBrL0PfDko51BRFUoPq4mXen57rGNmjxe7oVyAdCLlQ43"
#}

client = Client(
    "myGdLpz",  # consumer_key
    "7CyRE2o10a3ji7Or5QdkRET4mORPHpBcZdgVrv8l1pdEfqRIFv",  # consumer_secret
    "XLWbzoFAgX5Hdg0ViNS9pd55wYdH7THdIOaSVPaphtiONjnKSE02JagonMx6",  # token_key
    "0d5hQOhUuQ3U5oOmqs2Nbmy8lcBrL0PfDko51BRFUoPq4mXen57rGNmjxe7oVyAdCLlQ43"  # token_secret
)
uri, headers, body = client.sign("https://public.apps.ubuntu.com/download/net.launchpad.click-webapps/xkcd/net.launchpad.click-webapps.xkcd_5_all.click")

response = requests.get(uri, headers=headers)

with open("net.launchpad.click-webapps.xkcd_5_all.click", "wb") as f:
     f.write(response.content)
```

aleixpol, 2015-06-18:
@notmart maybe we want to integrate the applications directory into the home screen lookup path?

aleixpol, 2015-06-19:
So thing is it's already in the look-up path, even though it doesn't work at the moment, because it's using aa-exec-click. The only reason why it's not working is because there's a hook that modifies the desktop file to ensure it's going to be executed under apparmor.

Here it gets messy:
* We can disable the apparmor hook. Hooks can be edited with ```find /usr -name "*\.hook"``` and disabled, theoretically with ```click hook remove ubuntu-app-launch-desktop run-system```. Only that in practice it doesn't work, or I haven't been able to.
* We can fix apparmor so that applications get to be launcher properly. Now this sounds very reasonable, since we'll be leveraging an interesting technology anyway, in principle. I've spent about 3h trying to make it work, but didn't manage. What we need is to make sure the application can connect to ```/run/user/<pid>/wayland-0```, should be easy but I wasn't able to alter the behavior work.

aleixpol, 2015-06-19:
We did some progress, with the help of @apachelogger. Saving here, just in case I might flash my phone:
```
phablet@ubuntu-phablet:~$ cat /etc/apparmor.d/abstractions/wayland
owner /run/user/*/wayland-0 rw,
owner /run/user/*/weston-shared-* rw,
owner /tmp/wayland-shm-* rw,
```
Still, will need to keep whitelisting stuff.

aleixpol, 2015-06-19:
I just ran the first click-based application. That was the calculator, which is bundled with Ubuntu Phone.

Things to be done:
* Add the file specified on the comment before.
* ```echo "#include <abstractions/wayland>" >> /etc/apparmor.d/abstractions/base```
* prevent aa-exec-click from closing if there's some .X stuff.
```
phablet@ubuntu-phablet:/usr/share/click/preinstalled/com.ubuntu.calculator/2.0.182$ diff /usr/bin/aa-exec-click.original /usr/bin/aa-exec-click
54c54
<         exit 1
---
> #        exit 1
```

star-buck, 2015-06-20:
@apachelogger :
please find a way of shipping for next IMG and add small comment on how / where its done.

star-buck, 2015-06-20:
test app:
http://ubuntuphone.es/app-store/app/com.popey.stallboard

star-buck, 2015-06-20:
testing this from the es-appstore:
http://ubuntuphone.es/app-store/app/com.ubuntu.filemanager

apachelogger, 2015-06-23:
jon something for you to look at in the morning

we'll need the abstractions/wayland file packaged. that package probably also needs to dpkg-divert abstractions/base and add an include on wayland and divert aa-exec-click and somehow get rid of the exit.

it's slightly shite really.

alternatively for aa-exec-click we could probably fork the source and apt-hold it, I understand this hack is supposed to go away eventually anyway.

the apparmor abtractions stuff is here to stay though I fear.

star-buck, 2015-06-23:
@jriddell : can you have a look and try get that packaged + updated here:
http://mobile.kci.pangea.pub/pool/main/

jriddell, 2015-06-24:
plasma-phone-settings-packaging and plasma-phone-settings updated, waiting on mobile ci to reappear to update

star-buck, 2015-06-24:
Jonathan:
Can you check with Harald or have a look at the server yourself?
The server itself seems to be up and running fine, not sure if the :8080
pointed to another server or if something on the server needs to be started
for jenkins to come up again?

star-buck, 2015-06-24:
Should be fixed, waiting for jenkins to fully come up again...

star-buck, 2015-06-24:
triggered rebuilding:
- plasma-phone-settings
- plasma-phone-settings-packaging

notmart, 2015-06-24:
the new file is correctly here, but it still needs some files installed by upstream packages to be patched: how that should be done?

aleixpol, 2015-06-26:
@star-buck asked for a testing tool, this puts together what I had on the previous snippet to be used directly.
https://raw.githubusercontent.com/plasma-mobile/xutils/master/click-get

Use as: ./click-get <anyurl>

bhush9, 2016-03-01:
considering IMG27 ships click packages and they somewhat works.. we should close this

star-buck, 2016-03-02:
lets keep this open for info, its LTS, meaning Long Term Solution

closed 2016-03-02 by star-buck