blue-systems/plamo-initial-dev_DEPRECATED #27: [phone calls]: missing audio

star-buck, 2015-06-10:


davidedmundson, 2015-06-15:
Notes for Martin and packagers

Step 1: Making the audio in general work

Requires
plasma-phone-settings
plasma-phone-components

You should now hear a "boop" sound when the phone starts. 

(This isn't enough to fix audio calls)

Step 2:
remove the telepathy-ofono-mission-control plugin 

I replaced this with a startup script in 
in plasma-phone-settings. That conflicts with this plugin, because I didn't understand the plugin.

*IF* you install latest tp-qt you will need telepathy-ofono from this branch
https://code.launchpad.net/~akulichalexander/telepathy-ofono/telepathy-ofono to compile.

This should be enough to run ktp-call-ui from git with branch "davehack", and have audio calls work. 

Step 3:
Put this all into the existing UI. I've been fixing up a fork of 
https://github.com/nemomobile/voicecall

I have *not* finished this.

mck182, 2015-06-15:
So I've made the telepathy-ofono-mission-control plugin work, Telepathy now successfully (and automatically) lists an ofono account (and would list 2 accounts if there would be two simcards etc) with all the required properties.

I'm now going to try to get some simplified version of ktp-call-ui working with this account.

mck182, 2015-06-16:
Today's update - while investigating the ofono<->pulse integration, I did apt-get upgrade and ofono started to produce proper sound during calls, even the mic started working. So I've started to write a qml plugin which will make the dialer work with telepathy.

mck182, 2015-06-17:
Basic dialer functioning with Telepathy-ofono is now in plasma-phone-components branch mck182/tpCaller

The hangup however does not work, this seems to be a bug in telepathy-ofono not telling ofono to actually hang up. I'll have to read its source tomorrow to better understand it.

star-buck, 2015-06-18:
did you find out to make tpCaller working?
Once working, you"ll merge it into plasma-phone-componets master or consider split into its own app which then integrates with the module phonebook/telepathy-contacts/kpeople?

mck182, 2015-06-18:
Yes, I'm now able to make ongoing calls including hanging up.

Right now I'm working on being able to receive incoming calls as well, which is slightly more work. Once done, I'll merge to master of plasma-phone-components; it can be splitted into standalone thing but personally I don't see the need for it.

mck182, 2015-06-19:
Ok so I've merged my branch to plasma-phone-components master.

Current state:
 * outgoing calls all work (including speaker and mic)
 * incoming calls work with some issue on first incoming call
 -> the app is started by MissionControl, but MC never hands the channel to the application to handle it; MC prints a dbus error about never receiving a reply, so my guess would be some AppArmor issue. I couldn't figure it out today will try again. IF the app is started first, then incoming call works as expected (sometimes might take a while on first incoming call to react, not sure why yet)
 * incoming call rings
 * missed incoming call /should/ leave a notification, but for some reason does not seem to
 * qDebug output is written to ~/dialer.log so it can be retrieved even when the app is not started from console

What's needed to have it working:
 * David's pulseaudio config
 * exported FORCE_RIL_NUM_MODEMS=1 for now (part of bin/plasma-phone)
 * installed phonon4qt5-backend-gstreamer
 * installed kde-runtime-data (for ringing sound)
 * installed telepathy-ofono
 * installed telepathy-ofono-ril-mc-plugin
 
That should be all.

There's still quite some work needed on the qml<->telepathy glue, so I'll try to finish that on Monday

star-buck, 2015-06-19:
@apachelogger :
please package tpCaller/ rebuild plasma-phone-components with the pulseaudio config          from this ticket:
https://github.com/plasma-mobile/mobster/issues/66#issuecomment-113295443

@mck182 :
updating/packaging should install a tpCalller icon on the phone app section?

mck182, 2015-06-22:
@star-buck tpCaller was just the branch name, but the application is the same dialer application that was there before, which I think is the "Phone" app (but I was never able to execute any of them from the launcher).

I'll make sure it installs an icon there.

star-buck, 2015-06-23:
IMG 119:
typing a number and then pressing the "phone icon" to initiate a call makes the dialer quit.

star-buck, 2015-06-23:
found some missing packages as described above in IMG119:
- kde-runtime-data
- phonon4qt5-backend-gstreamer

Bute even after installing still crashes, so also check if those are in place:
- David's pulseaudio config
- exported FORCE_RIL_NUM_MODEMS=1 for now (part of bin/plasma-phone

mck182, 2015-07-03:
Freshly flashed img 127 works fine with outgoing calls and incoming calls afterwards. Update of plasma-phone-components makes incoming calls work just fine without previous opening of the dialer application too (updated package should end up on next image).

So please retest and close this if everything works fine.

---

I've experienced that sometimes the calls might not work, this is ofono behaving in a strange way and I'm not sure what's up; however calls can always be verified via ssh/shell/konsole with

qdbus --system org.ofono /ril_0 org.ofono.VoiceCallManager.Dial "your_number_here" ""

(the number must be in quotes then followed by space and two quotes; empty string)

and hanging up with

qdbus --system org.ofono /ril_0 org.ofono.VoiceCallManager.HangupAll

Rebooting the phone always made it work again. After some investigation this appears to be ofono being unable to read the sim info and so when a call is requested, it just errors out with "Operation failed". The phone shows up as having signal but the number is actually unavailable when called. Might be some modem driver related issue.

star-buck, 2015-07-21:
seems to work here with IMG 135, closing this for now.

closed 2015-07-21 by star-buck