blue-systems/plamo-initial-dev_DEPRECATED #36: [IMG]; add space to make system partition bigger

star-buck, 2015-06-11:


jriddell, 2015-06-12:
using dd from /dev/zero made gzip go OOM on the image untar part of process
using dd from /dev/urandom made thrift server crap out
using fallocate of 1GB size made gzip OOM 
using fallocate of 10x100MB size made phone not boot
so I guess I need to experiment with different sizes of file to see what doesn't break

star-buck, 2015-06-24:
any idea to increase the / partition?
as of IMG119 it looks like this:
'''/dev/loop0                      2.0G  1.6G  276M  86% /
/dev/mmcblk0p28           13G  3.4G  8.6G  29%  /home'''

jriddell, 2015-07-16:
adding empty fallocate files doesn't do anything, they just take up space without making the device bigger, we need to work out what is making that loopback device and how to make it bigger by default

star-buck, 2015-07-21:
this has become a problem now, one cannot install anything else nor upgrade as of current IMG (135):
[IMG]: root /-partition is 100% full

We need to find a method to make more space on the partition after reboot.

star-buck, 2015-09-19:
any new ideas on this?

bhush9, 2015-09-19:
One thing I noticed lately are apt archives.. When we upgrade with apt downloaded files are not cleaned.. Which turns out to be wasted space.. Can we somehow ship some config so that archives are deleted when upgrades are finished..??

bhush9, 2016-01-19:
So, I researched bit about how to get more space initially, I realized its system-image-upgrader in the  recovery.img which defines the initial rootfs space.. I've created a recovery.img which would create rootfs of 4GB instead of 2GB default.

To use that,

```
wget http://neon.plasma-phone.org/recovery.img
ubuntu-device-flash --server="http://neon.plasma-phone.org/" touch --channel="neon-mobile/devel-proposed" --bootstrap --recovery-image recovery.img --developer-mode --password 1234
```

Here I added *--recovery-image recovery.img* in older command. If you get error like,

```
unknown flag `recovery-image'
```

You will need newer version of ubuntu-device-flash, for that.

```
sudo add-apt-repository ppa:ubuntu-sdk-team/ppa
sudo add-apt-repository ppa:phablet-team/tools 
sudo apt-get update; sudo apt-get install ubuntu-device-flash
```

bhush9, 2016-01-20:
Actually ignore that... that recovery image is somehow broken and doesn;t work... :-(

bhush9, 2016-06-29:
Since we switched stack and now we no longer use dedicated system.img/ubuntu.img (ubuntu touch mechanism) file as rootfs this is no longer relevant. Closing this.

closed 2016-06-29 by bhush9