blue-systems/plamo-initial-dev_DEPRECATED #52: [usb] connect and get access to file system (via mtp)

star-buck, 2015-06-13:


aleixpol, 2015-06-13:
Just researched a bit, I saw there's already an mtp server in our devices: https://launchpad.net/mtp
It doesn't start though, that's the error message, will need to look into it further:

phablet@ubuntu-phablet:~$ /usr/bin/mtp-server 
E0613 21:32:39.052184  2359 server.cpp:441] Could not start the MTP server:org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.UnityGreeter was not provided by any .service files

star-buck, 2015-06-14:
cloned our own version: https://github.com/plasma-mobile/mtp-server

star-buck, 2015-06-15:
@jriddell : 
replace mtp-server package with ods package from here: https://github.com/plasma-mobile/mtp-server

vHanda, 2015-06-15:
The `mtp-server` command needs to be manually run.

    $ adb shell
    > $ export DISPLAY=:0
    > $ mtp-server

jriddell, 2015-06-15:
mtp-server does not currently run as server/server.cpp requires unity greeter
"Could not start the MTP server:org.freedesktop.DBus.Error.ServiceUnknown: The name com.canonical.UnityGreeter was not provided by any .service files"
also mtp-state is an upstart job to decide on whether it can start mtp-server and it needs env var persist.sys.usb.config=mtp which isn't currently set.  that needs either set or known that we can ignore it

aleixpol, 2015-06-15:
The UnityGreeter error is what Vishesh was workarounding, AFAIU. Maybe you're not running his patch?

jriddell, 2015-06-15:
mm yes.. investigating..

jriddell, 2015-06-15:
mtp-server.conf simplified should work now, adding to rebuild queue..

jriddell, 2015-06-15:
built

star-buck, 2015-06-16:
doesnmt work as of 104

star-buck, 2015-06-17:
v105: still not working

star-buck, 2015-06-17:
107: package installed, but still not started:
The `mtp-server` command needs to be manually run.

    $ adb shell
    > $ export DISPLAY=:0
    > $ mtp-server

star-buck, 2015-06-18:
works as of 109! :)

closed 2015-06-18 by star-buck