blue-systems/plamo-initial-dev_DEPRECATED #58: [mobster]: specific package listings to build IMG+1

star-buck, 2015-06-14:
101: boots fine

star-buck, 2015-06-14:
102: first try: did hang at google logo

star-buck, 2015-06-15:
for IMG 103:
- rebuild simplelogin (should start maliit)
- rebuild plasma-maliit-plugins

star-buck, 2015-06-15:
for IMG 104:
- rebuild simplelogin
- use updated mtp-server package from ods
The `mtp-server` command needs to be manually run.

    $ adb shell
    > $ export DISPLAY=:0
    > $ mtp-server

star-buck, 2015-06-15:
this is ongoing, so ticket never closes :)

jriddell, 2015-06-15:
new mtp-server is in

star-buck, 2015-06-17:
for IMG 106:
* rebuild plasma-phone-components
- mtp-server from here is still not in: http://dci.pangea.pub/plasma-phone/debian/pool/main/m/mtp/

jriddell, 2015-06-17:
Image 107 is up with new mtp-server (but not yet plasma-phone-components)

star-buck, 2015-06-18:
for image 110:
- add koko package from ci packages so its installed default

closed 2015-06-23 by star-buck