blue-systems/plamo-initial-dev_DEPRECATED #59: [top panel]: pulling down sometimes crashes

star-buck, 2015-06-14:


mgraesslin, 2015-06-14:
I was able to catch an assert in libhybris. It might be related:
kwin_wayland: hwcomposer_window.cpp:274: void HWComposerNativeWindow::lockFrontBuffer(HWComposerNativeWindowBuffer**): Assertion `m_frontBuf->busy == 2' failed.
Application::crashHandler() called with signal 6; recent crashes: 1
QSocketNotifier: Invalid socket 9 and type 'Read', disabling...

Probably we need to spend some more time in trying to understand hwcomposer and get that up correctly. Currently I think it's still too much copy'n'paste from the libhybris demo.

mgraesslin, 2015-06-19:
I haven't seen this crash in a while - is it still happening?

star-buck, 2015-06-20:
seems fixed for now

closed 2015-06-20 by star-buck