blue-systems/plamo-initial-dev_DEPRECATED #65: [dekko]: build into .deb package

star-buck, 2015-06-15:
https://code.launchpad.net/~ubuntu-touch-coreapps-drivers/+recipe/dekko-daily

star-buck, 2015-06-15:
code/recipe: https://code.launchpad.net/~ubuntu-touch-coreapps-drivers/+junk/dekko-packaging

apachelogger, 2015-06-15:
old version from before it was converted to a click. either we enable using clicks or dekko needs substantial changes to the cmake code to get it back into proper system paths

the old version appears to do something semi-resaonable though, so here is that:
https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5142819/+listing-archive-extra

it has a slight modification over the actual build as the desktop file referenced a wrong exec path

boudewijnrempt, 2015-06-16:
Okay... So that's how I killed my laptop on Monday morning?

boudewijnrempt, 2015-06-18:
Ok -- so I'll check and see if I can make debs out of the latest version as well? Or do we form from the old version and go from there?

star-buck, 2015-06-18:
latest version

closed 2015-06-23 by star-buck