blue-systems/plamo-initial-dev_DEPRECATED #66: [audio] enable pulseaudio by default

star-buck, 2015-06-15:


davidedmundson, 2015-06-15:
Done needs packaging.

star-buck, 2015-06-17:
which package names? normal CI or ODS?

star-buck, 2015-06-17:
test sound with this app:
https://uappexplorer.com/app/com.ubuntu.developer.robert-ancell.animal-farm

to install on phone, adb shell in and execute:

sudo apt-add-repository ppa:plasma-phone/ubuntu-apps
sudo apt-get update
sudo apt-get install animal-farm

star-buck, 2015-06-18:
How can we test if volume is up and sound should be generally working?

davidedmundson, 2015-06-18:
in adb shell run
 paplay /usr/share/sounds/ubuntu/ringtones/Ubuntu.ogg

star-buck, 2015-06-18:
just tried this on the latest 109 IMG: still silence :(
@mck182 : any way we can make this working default?

mck182, 2015-06-18:
I thought this is done by David's pulse audio config, which is what made it working here. I'm not sure if this was already packaged and present on the image however.

https://github.com/plasma-mobile/plasma-phone-settings/issues/1

star-buck, 2015-06-19:
@apachelogger :
Add file below as /etc/pulse/default.pa (overwrite file shipped by pulseaudio) :
https://github.com/plasma-mobile/plasma-phone-settings/blob/master/etc/pulse/default.pa

star-buck, 2015-06-19:
link is dead, so restore config file because we still need it?:
https://github.com/plasma-mobile/plasma-phone-settings/commit/b9f057cf19f80d027c650445b2a237beaa29a4ff

star-buck, 2015-06-20:
for testing:
in adb shell run
paplay /usr/share/sounds/ubuntu/ringtones/Ubuntu.ogg 

sudo add-apt-repository ppa:plasma-phone/ubuntu-apps
sudo apt-get update
sudo apt-get install animal-farm

star-buck, 2015-06-20:
to make animal-farm have sounds, this is required:
sudo apt-get remove qtubuntu-media

jriddell, 2015-06-23:
mobster committed:
commit 8e885c9e649cbf23ae749f6556f1038b8642c8fe
Author: Jonathan Riddell <jr@jriddell.org>
Date:   Tue Jun 23 12:46:59 2015 +0200

    enable audio easier by removing qtubuntu-media and adding gstreamer0.10-fluendo-mp3

closed 2015-06-23 by jriddell