blue-systems/plamo-initial-dev_DEPRECATED #72: [packaging]: patched qdeclarative in CI

notmart, 2015-06-17:
the qtdeclarative package needs the following patch to fix some touch event coordinates issues:
https://codereview.qt-project.org/#/c/114424/

jriddell, 2015-06-23:
building now in PPA https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+packages

jriddell, 2015-06-23:
qemu failure, I knew there was a reason we don't use launchpad PPAs for building arm.  trying again in on-demand...

star-buck, 2015-06-23:
Hey Jonathan,

with on-demand, you build them ending in our new repo?
http://mobile.kci.pangea.pub/pool/main/

As I understood, we now have all kci mobile packages there with the
possibility of selectively updating single ones on-demand...

Greetings, Clemens.

jriddell, 2015-06-23:
well it compiled on delta ~/pbuilder/vivid_result/qtdeclarative-opensource-src_5.4.1-1ubuntu5kubuntu1.dsc but as on-demand is now gone it's not much use, I'll work out how to use mobile kci tomorrow

star-buck, 2015-06-24:
@jriddell : ok, thanks!

jriddell, 2015-06-24:
waiting for mobile ci to reappear

star-buck, 2015-06-24:
@jriddell : mobile ci reappeared

jriddell, 2015-06-24:
publishing on mobile ci needs ssh access to the publisher@ account which I don't think I have

star-buck, 2015-06-24:
@jriddell : try again login to mobile jenkins with "jriddell", you should have adminstrative privileges now... if not, I probably need to restart server.

jriddell, 2015-06-25:
got it in  http://mobile.kci.pangea.pub/pool/main/q/qtdeclarative-opensource-src/

closed 2015-06-25 by jriddell