blue-systems/plamo-initial-dev_DEPRECATED #75: [apps]: muon mobile for installing apps with gui

star-buck, 2015-06-18:


aleixpol, 2015-06-19:
Status update:
* Application starts but needs work to adapt to the form factor.
* QApt works, although the UI is widget-based and doesn't look very well.
* We need to investigate whether there's anything we can do about the PackageKit click backend, because it seems to work but not for some important actions
 * Listing and install-local works, install and remove doesn't
* We need to see what to do with the app-install desktop files, whether we can adapt it for phone. Or finally move to appstream for good.

star-buck, 2015-06-20:
installing "muon" package currently still drags in lots of (qt4-)dependencies

aleixpol, 2015-06-20:
We looked into it with @shadeslayer. It seems to be because of kdesudo (at least). This case is solved in master.

star-buck, 2015-06-25:
@apachelogger : please package muon from master and include in next IMG

star-buck, 2015-06-25:
needs additional package to work:
* app-install-data

apachelogger, 2015-06-26:
@aleixpol which package do we want muon or muon-discover? (I presume latter)

aleixpol, 2015-06-26:
@apachelogger muon-discover

apachelogger, 2015-06-26:
built and add to mobster

@jriddell kindly note muon-discover on next image

jriddell, 2015-06-30:
it's on.

star-buck, 2015-07-01:
126: icon is there, but muon doesnt start

star-buck, 2015-07-21:
still doesnt start in IMG 135.
Either remove the app completely or make it work until akademy.

aleixpol, 2015-07-21:
It's a weird issue, if it's started from adb, it runs correctly, from plasma it doesn't start. Anybody knows how to debug it?

star-buck, 2015-07-21:
maybe use konsole shipped on the IMG and start "muon-discover" from there?

aleixpol, 2015-07-21:
I realized it's a small issue with Plasma QtQuick Controls, I'm trying to figure it out. Using Konsole was useful, thanks @star-buck :D

aleixpol, 2015-07-21:
Should be fixed now, in master. I'm seeing different visual issues as I'm testing it, will look into them, but it's ready for testing.

star-buck, 2015-07-31:
IMG 137: still doesnt start :/

aleixpol, 2015-07-31:
Do we have an up to date plasma-framework?

aleixpol, 2015-08-04:
Just looked into it, we need to get a fresh Muon Discover.

More precisely, we need 9352fbd124c0f1e082b1d39620d95d73d239215e from the 21st July, currently we have a version for the 17th July.

@apachelogger Can we maybe get Muon in the CI?

apachelogger, 2015-08-05:
@aleixpol http://mobile.kci.pangea.pub:8080/job/vivid_unstable_muon_src/ ?

jriddell, 2015-08-05:
new version building now

closed 2015-08-05 by jriddell