blue-systems/plamo-initial-dev_DEPRECATED #76: [apps]: music player app

star-buck, 2015-06-18:


bhush9, 2016-01-08:
img17 have ubuntu's music-app installed.

bhush9, 2016-07-27:
Ubuntu's music app will not work in recent images due to incompatibility with libhybris. So one will need to find something else as solution of music player app

star-buck, 2017-12-09:
any suggestions what to ship for now?

bhush9, 2017-12-10:
We can try elisa / babe music player..

luebking, 2017-12-10:
mpd. It has like no footprint, can maintain a local database, proxy a remote one and there's a truckload of clients.

star-buck, 2017-12-13:
for the purpose of demonstrating "a full linux stack + plasma desktop in your pocket" awesomeness
- install "audacious"
- set default to "Winamp Style" and "double size"

bhush9, 2017-12-14:
I might be blind, but i can't find option to set "double size"?

star-buck, 2017-12-14:
Its in one of the dropdown menus, not the settings.

llelectronics, 2017-12-14:
If you right click on the player you can set it.

bhush9, 2017-12-14:
> If you right click ...

And how do I do it on phone? :P

star-buck, 2017-12-14:
Just see on the desktop and ship the config.

bhush9, 2017-12-14:
Yeah, figured out config entry.. will build image in few minutes.

bhush9, 2017-12-14:
Image `20171214-165756` includes audacious, however somehow the Winamp style is not default, will correct in next image.

star-buck, 2017-12-15:
lets also set "default font sizes" for playlist and player from 9 to 24 pixels, and disbale bitmap font.

closed 2017-12-15 by star-buck