blue-systems/plamo-initial-dev_DEPRECATED #82: [taskmanager]: port libtaskmanager to kwayland

star-buck, 2015-06-19:


mgraesslin, 2015-06-19:
Relevent task on todo.kde.org: https://todo.kde.org/?controller=task&action=show&task_id=558

eikehein, 2015-06-19:
I set up a Wayland dev setup on an Intel system earlier this week after determining that my nVidia workstation isn't usable for the purpose and started looking into adding kwayland support to the lib.

eikehein, 2015-06-24:
I've concluded that libtaskmanager as it exists right now is virtually unportable. It's code is heavily tied to KWindowSystem API that isn't supportable on Wayland (based around NetWM-isms and X11 WIds), along with lower-level X11-specific bits like NETWinInfo and NETRootInfo. Bifurcating everything to use either KWS or kwayland would turn the lib into a complete mess, and amounts to shoving a full second implementation in there.

Instead, I've just pushed a newly-written PlasmaWindowModel to kwayland and ported the TaskSwitcher in plasma-phone-components to it. The model is basic, but exposes everything kwayland's PlasmaWindowManagement can do at present, and it's enough to make the TaskSwitcher work. (In lieu of window type support, the frontend is currently filtering on non-empty window titles.)

In the longer-term, the idea is to design a new KWindowSystem API that can be supported across more different windowing systems, and add a model to KWindowSystem. This model can then either expose X11 data or proxy the model in kwayland. libtaskmanager will then have nearly all of its C++ API removed (which mostly just duplicates KWS/kwayland anyway, and has no users anymore since all applets use the TasksModel built on top of it), and everything libtaskmanager actually adds to the equation (sorting, grouping, launchers) will be implemented as a proxy model on top of the KWS one.

bhush9, 2016-07-26:
I guess this can be closed?

star-buck, 2016-07-26:
@bhush9 : is it fixed? discarded? why?

star-buck, 2016-07-26:
i mean is it working in relation to plasma mobile, that is

bhush9, 2016-07-26:
yeah, it is fixed, on mobile we are using libtaskmanager-ng based backend from quite long time.. and 5.7 release of plasma also uses new backend.

bhush9, 2016-07-26:
Though I am not sure how fully functional wayland backend is for taskmanager, that is why my question.

star-buck, 2016-07-26:
is latest pm img using this, then it can be closed, even if buggy thats other tickets then

eikehein, 2016-07-26:
The Wayland backend does everything the PM TM needs and Marco ported it, feel free to close :)

star-buck, 2016-07-26:
and closed

closed 2016-07-26 by star-buck