blue-systems/plamo-initial-dev_DEPRECATED #87: [apps]: okular pdf reader

star-buck, 2015-06-19:


apachelogger, 2015-06-23:
@jriddell https://github.com/plasma-phone-packaging/okular/tree/kubuntu_unstable

install files need updating otherwise it should be good. perhaps you can twiddle the installs tomorrow morning

star-buck, 2015-06-25:
@apachelogger : is okular package suppoed to land here: http://mobile.kci.pangea.pub/pool/main/ ?

luebking, 2015-06-25:
Sidenote: you likely want https://github.com/xylosper/okular-backend-mupdf (originally written by Pino Toscano) because poppler performance sucks. Big time.

apachelogger, 2015-06-26:
@star-buck yes

star-buck, 2015-06-30:
@apachelogger @jriddell :
any update on packaging this?

jriddell, 2015-07-02:
building now in on-demand..

jriddell, 2015-07-02:
up at http://mobile.kci.pangea.pub/pool/main/o/okular/
harald should also add it to mobile kci (or teach me how to do so)

closed 2015-07-02 by jriddell