blue-systems/plamo-initial-dev_DEPRECATED #88: [LTS]: game emulators

star-buck, 2015-06-19:
maemo mame emulator: 
http://www.my-maemo.com/software/applications.php?fldAuto=1069&faq=46

n900 n64:
https://www.youtube.com/watch?v=dPiaoE-Eixw

mgraesslin, 2015-06-19:
In general all X game emulators should just work (TM) assuming we install breeze window decoration

star-buck, 2015-06-19:
@apachelogger:
enable "multiverse" in sources.list

package to be installed on IMG:
- kwin-style-breeze

star-buck, 2015-06-21:
N9 EmuMaster:
https://bitbucket.org/elemental/emumaster/src

star-buck, 2015-06-23:
N9 QML Emulator with keyboard simulation:
https://bitbucket.org/elemental/emumaster/src/eb5048c7b575745668ddbd8d2bfca3e69cf8b035?at=default

star-buck, 2015-06-24:
Meego Touch Frameworks heritage:
https://en.wikipedia.org/wiki/MeeGo

star-buck, 2015-06-24:
Source Code:
http://code.openhub.net/project?pid=Ehp4c7MVUyI&prevcid=1&cid=LYFNVWBqFSY&fp=28680&projSelected=true