blue-systems/plamo-initial-dev_DEPRECATED #101: [phonecalls] dialer layout broken

sebasje, 2015-06-20:
The letters are weirdly overlapping the number in the dialpad, it seems the layout needs corrections.

mck182, 2015-06-22:
@notmart are you going to continue your work on the dialer UI? Or should I take that over completely?

mck182, 2015-06-23:
@notmart Here's a patch fixing the layout - basically it moves the buttons in the RowLayout below the GridLayout into the GridLayout itself; that fixes the overall width and everything's fine.

For some reason though the number 4 will not appear unless the max point size is set to 22, I have spent about an hour investigating and can't figure out why the heck that 4 wouldn't appear and I still don't know, so I left it at 22 for now (the size was 23~24 before anyway). It still feels a bit unbalanced because of the letters next to the numbers, but there's nothing to do about that unless we move the letters below the numbers.

Anyways, here's the patch, if you're ok with it, I'll commit.

```diff
 dialer/package/contents/ui/Dialer/DialPage.qml     | 16 ++++-----
 dialer/package/contents/ui/Dialer/Dialer.qml       | 38 ++--------------------
 .../package/contents/ui/Dialpad/DialerButton.qml   |  9 ++---
 dialer/package/contents/ui/Dialpad/Dialpad.qml     | 33 ++++++++++++++++++-
 dialer/package/contents/ui/main.qml                |  2 +-
 5 files changed, 47 insertions(+), 51 deletions(-)

diff --git a/dialer/package/contents/ui/Dialer/DialPage.qml b/dialer/package/contents/ui/Dialer/DialPage.qml
index 3e0c252..7e7bf34 100644
--- a/dialer/package/contents/ui/Dialer/DialPage.qml
+++ b/dialer/package/contents/ui/Dialer/DialPage.qml
@@ -24,12 +24,12 @@ import org.kde.plasma.components 2.0 as PlasmaComponents
 
 Column {
     spacing: 0
+    anchors.fill: parent
+
     PlasmaComponents.TabGroup {
-        anchors {
-            left: parent.left
-            right: parent.right
-        }
         height: parent.height - tabbar.height
+        width: parent.width
+
         History {
             id: history
         }
@@ -43,10 +43,8 @@ Column {
     PlasmaComponents.TabBar {
         id: tabbar
         height: units.gridUnit * 5
-        anchors {
-            left: parent.left
-            right: parent.right
-        }
+        width: parent.width
+
         tabPosition: Qt.BottomEdge
         PlasmaComponents.TabButton {
             iconSource: "view-history"
@@ -64,4 +62,4 @@ Column {
             tab: dialer
         }
     }
-}
\ No newline at end of file
+}
diff --git a/dialer/package/contents/ui/Dialer/Dialer.qml b/dialer/package/contents/ui/Dialer/Dialer.qml
index d6760e9..6d9ec77 100644
--- a/dialer/package/contents/ui/Dialer/Dialer.qml
+++ b/dialer/package/contents/ui/Dialer/Dialer.qml
@@ -42,11 +42,13 @@ Item {
             fill: parent
             margins: units.largeSpacing
         }
+
         PlasmaComponents.Label {
             id: status
             Layout.fillWidth: true
             Layout.minimumHeight: parent.height / 6
             Layout.maximumHeight: Layout.minimumHeight
+
[... 109 lines cut ...]
+
+        enabled: status.text.length > 0
+        opacity: enabled ? 1 : 0.5
+        source: "edit-clear"
+        callback: function(text) {
+            if (status.text.length > 0) {
+                status.text = status.text.substr(0, status.text.length - 1);
+            }
+        }
+    }
+}
diff --git a/dialer/package/contents/ui/main.qml b/dialer/package/contents/ui/main.qml
index 618fb6b..13ca450 100644
--- a/dialer/package/contents/ui/main.qml
+++ b/dialer/package/contents/ui/main.qml
@@ -29,7 +29,7 @@ ApplicationWindow {
     id: root
 
 //BEGIN PROPERTIES
-    width: 600
+    width: 1080
     height: 800
 
     //keep track if we were visible when ringing
```

star-buck, 2015-06-23:
@mck182 :
Let's move the letters below the numbers.

mck182, 2015-06-24:
All done in master, please rebuild plasma-phone-components.

star-buck, 2015-06-24:
triggered rebuild of plasma-phone-components

star-buck, 2015-06-24:
fixed.

closed 2015-06-24 by star-buck