blue-systems/plamo-initial-dev_DEPRECATED #104: [packaging] updated kinfocenter needed for energy module

sebasje, 2015-06-20:
The kinfocenter package is still at 5.2.2, 5.3 contains a module for energy consumption of individual processes, we'd like to ship this on the phone (perhaps with an adjusted UI). First step is to get a new kinfocenter on it, so we can improve the UI of the energy module (which will be loaded into the settings app).

jriddell, 2015-06-24:
needs added to mobile ci which is currently down

jriddell, 2015-06-24:
powerdevil bulit yesterday in mobile ci http://mobile.kci.pangea.pub/pool/main/p/powerdevil/

jriddell, 2015-06-24:
wrong bug, still open

jriddell, 2015-06-24:
I don't think I have privilages to add new jobs to jenkins

star-buck, 2015-06-24:
@jriddell: you now should have admin rights, please try again if it works.

jriddell, 2015-06-24:
logging out and in doesn't seem to give me any new powers

star-buck, 2015-06-24:
@jriddell : what login you use, jriddell or "jr"?

jriddell, 2015-06-25:
harald added kinfocentre jobs to mobile kci which are running now

apachelogger, 2015-06-26:
built http://mobile.kci.pangea.pub:8080/job/vivid_unstable_kinfocenter_pub/

apachelogger, 2015-06-26:
@jriddell  I also added it to mobster, so the next image should have it

closed 2015-06-30 by jriddell