blue-systems/plamo-initial-dev_DEPRECATED #106: Implement brightness changing in powerdevil

sebasje, 2015-06-20:
Has to go through KAuth and /sys/class/backlight/*/brightness

bhush9, 2016-08-08:
Started putting patches on phab which I had worked on earlier,

https://phabricator.kde.org/D2369
https://phabricator.kde.org/D2372

Once last patch is in, will put patch for hwcomposer backend on review..

bhush9, 2016-08-11:
[image]

voila.. now battery monitor at least shows brightness slider.. but that doesn't actually set brightness, this needs investigation

bhush9, 2016-08-11:
On more investigation this shows that powerdevil is currently reading brightness from ```/sys/class/backlight/lm3630/brightness``` and trying to write brightness on same interface however, this seems to be some dummy read-only interface and is not actually used by android lights module. Android lights module uses ```/sys/class/leds/lcd-backlight/brightness``` sysfs interface.

So to deal with this plan is,

- [ ] Adjust powerdevil code to also read value through /sys/class/leds/lcd-backlight/brightness
- [ ] Add dbus interface in KWin hwcomposer backend to write brightness value
- [ ] Add code in powerdevil to use dbus interface provided by KWin

bhush9, 2016-08-11:
On second thought.. this is still not making much sense and needs brainstorming.

Powerdevil have backlight helper which gets/sets the brightness through sysfs interface ```/sys/class/backlight/``` and it excepts type file in each subdir with raw, firmware or platform as content. Now ```/sys/class/leds/lcd-backlight/``` is as well similar brightness interface but it doesn't expose the type. Which is most likely broken driver or limitation, given that is led subsystem.

Now first option was to adjust backlight helper to read values from ```/sys/class/leds/lcd-backlight``` but the code highly depends on being valid type, so mostly adjusting this code doesn't make much sense.

Another option at hand was add different helper for hwcomposer but then all android devices doesn't use lcd-backlight interface.. (yet to confirm completely..). So this is again meh, and abstraction over this (lights HAL module) just supports setting brightness not reading.

Also with more search it seems lcd-backlight is interface specific to MSM/Snapdragon.

bhush9, 2016-08-12:
With https://phabricator.kde.org/D2419 powerdevil is adjusted to use ```/sys/class/leds/lcd-backlight``` interface instead of ```/sys/class/backlight/lm3630```. Now we ought to think solution for KWin hwcomposer backend resetting the brightness to 100% when toggling screen off-on..

Also I adjusted phone to use normal battery monitor,

- https://phabricator.kde.org/D2411
- https://phabricator.kde.org/D2412

bhush9, 2016-08-12:
https://phabricator.kde.org/D2419 had minor bug which made it to take any leds from leds subclass.. http://commits.kde.org/powerdevil/62a88ee30 fixed that.

bhush9, 2016-08-12:
After changes all of above, battery monitor is able to adjust brightness but as mentioned in previous comment, KWin adjust brightness to 100% when screen is turned off and on, for this two things will need to be adjusted

* http://phabricator.kde.org/D2424 - This makes kwin to not change brightness on its own
* change in powerdevil DPMS action to set brightness to 0 when DPMS is turned off.

After that brightness control is properly fixed..

bhush9, 2016-08-17:
this morning I had another idea that what if I listen to brightnessChanged signal from solid dbus service, and use it to track screen brightness. I did implement code for it.. but turnsout brightnessChanged is not getting emitted.

seriously this brightness thingie is can of worms..

bhush9, 2016-08-17:
In kwin idea above was implemented with, https://phabricator.kde.org/D2468 and powerdevil needs following fix for it to work properly : https://phabricator.kde.org/D2470

bhush9, 2016-08-18:
Hurr durr.. all changes are in.. today's image will have this included... \o/

bhush9, 2016-08-19:
20160819-141814 image have this working... great will be to have ambient light sensors working.. but thats different issue..

bhush9, 2017-12-10:
Can we close this ticket? it seems to have been resolved.

star-buck, 2017-12-10:
and closed

closed 2017-12-10 by star-buck