blue-systems/plamo-initial-dev_DEPRECATED #89: HWComposer

davidedmundson, 2015-06-20:
- kwin tears when scrolling

star-buck, 2015-06-25:
- shutoff screen completely when pressing power buttton
should be done in near future to not drain battery when phone is off powercable

mgraesslin, 2015-06-25:
> shutoff screen completely when pressing power buttton

it's actually not doable in hwcomposer, thus it becomes more complicated. It needs to be passed to powerdevil to do the normal screen brightness adjustment combined with a hook back to kwin to stop rendering at the same time.

As far as I can see this sounds like we need quite some work first:
* kglobalacceld port to wayland (also needed for volume buttons)
* integration in powerdevil
* hooks inside kwin

So it's not something which can easily be done and needs also discussion with the powerdevil developers on how to do the kwin/powerdevil integration properly (we need similar architecture for DPMS on DRM backend).

mgraesslin, 2015-06-26:
> kglobalacceld port to wayland (also needed for volume buttons)

review requests for kglobalaccel and kwin are prepared.

star-buck, 2015-09-19:
if klobalaccel and kwin are done, we could close this ticket and refer to "turn of screen" ticket for powerdevil?

mgraesslin, 2015-09-19:
the initial bug report also mentions the tearing problem which is not yet fixed.

star-buck, 2015-11-26:
screen turns off with img-12, so closing this

closed 2015-11-26 by star-buck