blue-systems/plamo-initial-dev_DEPRECATED #90: [packaging] package bluez5

sebasje, 2015-06-20:
In order to use bluedevil and its fancy plasmoid, we need bluez5, which isn't available in the ubuntu repos.

Debian has a package, perhaps we can import this? 

https://packages.debian.org/jessie/bluez

Harald said, we'll also need our own pulseaudio build.

jriddell, 2015-06-23:
packaging is in https://launchpad.net/~ubuntu-desktop/+archive/ubuntu/transitions just needs put in mobile kci

jriddell, 2015-06-25:
copied over bluez and pulseaudio to ppa:plasma-phone/ppa, bluedevil and bluez-qt needs added to mobile kci

jriddell, 2015-06-25:
<mck182> Riddell: apt-get upgrade always gets stuck on Setting up bluez (5.23-1ubuntu1~mtrudel1) ... here

more testing needed

mck182, 2015-06-25:
after it fails, running "sudo dpkg --debug=73773 --configure -a" gives this output:

(it starts with processing ubuntu-system-settings:armhf for which bluez seems to be a dep?)

D000001: process queue pkg bluez:armhf queue.len 1 progress 2, try 1
D000040: checking dependencies of bluez:armhf (- <none>)
D000400:   checking group ...
D000400:     checking possibility  -> libc6
D000400:       checking non-provided pkg libc6:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> libdbus-1-3
D000400:       checking non-provided pkg libdbus-1-3:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> libglib2.0-0
D000400:       checking non-provided pkg libglib2.0-0:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> libreadline6
D000400:       checking non-provided pkg libreadline6:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> libudev1
D000400:       checking non-provided pkg libudev1:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> init-system-helpers
D000400:       checking non-provided pkg init-system-helpers:all
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> lsb-base
D000400:       checking non-provided pkg lsb-base:all
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> kmod
D000400:       checking non-provided pkg kmod:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> udev
D000400:       checking non-provided pkg udev:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000400:   checking group ...
D000400:     checking possibility  -> dbus
D000400:       checking non-provided pkg dbus:armhf
D000400:       is installed, ok and found
D000400:     found 3
D000400:   found 3 matched 0 possfixbytrig -
D000040: ok 2 msgs >><<
D000040:     checking Breaks
Setting up bluez (5.23-1ubuntu1~mtrudel1) ...
D020000: trigproc_activate_packageprocessing pkg=bluez:armhf
D000002: fork/exec /var/lib/dpkg/info/bluez.postinst ( configure 4.101-0ubuntu25 )

...and then stays here forever.

jriddell, 2015-06-25:
It hangs on starting bluetooth in init.d script
/etc/init.d/bluetooth

jriddell, 2015-06-25:
building a new image to see if it can successfully install bluedevil and bluez, however it's an old bluedevil version because mobile kci doesn't yet include it

jriddell, 2015-06-25:
an unexpected build error, I'll need to investigate it 
"bluedevil : Depends: obex-data-server but it is not going to be installed"

star-buck, 2015-06-25:
tried to sudo apt-get update package , but never configs right, so breaks update process...

apachelogger, 2015-06-26:
bluez-qt and bluedevil in CI now

jriddell, 2015-06-26:
bluez needs rebuilt for a library transition, building now

jriddell, 2015-06-26:
image 121 is up with bluedevil in it, I've not booted it yet

jriddell, 2015-06-30:
booted image 122, it's on

closed 2015-06-30 by jriddell