blue-systems/plamo-initial-dev_DEPRECATED #94: [LTS]: reduce power consumption

star-buck, 2015-06-20:


sebasje, 2015-06-20:
Apparently, the third of the four cores is using the performance governor, which means it'll always run at full speed and not slow down, even when idle. This means it eats power even when not needed.

root@ubuntu-phablet:/sys/devices/system/cpu/cpu2/cpufreq# echo ondemand > scaling_governor
root@ubuntu-phablet:/sys/devices/system/cpu/cpu2/cpufreq# cat scaling_governor 
ondemand

This should be done somewhen during bootup.

Weirdly (hardware-weird!), the fourth core doesn't have cpufreq scaling, so we can't read or set it for the fourth core. Making the 3rd core ondemand should already make a difference to power consumption.

sebasje, 2015-06-20:
Found this document, which has some (more or less) useful information about power management in the hammerhead (Nexus 5) kernel.

https://github.com/atilag/hammerhead-nexus5-kernel/blob/master/Documentation/ABI/testing/sysfs-devices-system-cpu

sebasje, 2015-06-20:
Also, it may be interesting to try updating to this (newer) kernel, it seems to come from Cyanogen, and it's the Lollipop kernel, I believe we're using the Icecream sandwich kernel right now.

bhush9, 2016-07-27:
FWIW, we are now using 5.1 kernel.. (cm13) with newer stack, I don't fully understand what to change here? @sebasje can you expand bit on this?

sebasje, 2016-07-27:
root@ubuntu-phablet:/sys/devices/system/cpu/cpu2/cpufreq# echo ondemand > scaling_governor
root@ubuntu-phablet:/sys/devices/system/cpu/cpu2/cpufreq# cat scaling_governor 
ondemand

The above, this sets the CPU frequency scaling to something more power-friendly, it increases the speed as needed, but scales the CPU down when possible as to save energy.

bhush9, 2016-07-27:
With cm based stack all of the cores are using interactive mode,

```
root@hammerhead:/sys/devices/system/cpu # cat cpu{0,1,2,3}/cpufreq/scaling_governor                                                                                                              
interactive
interactive
interactive
interactive
```

Which from [documentation](https://android.googlesource.com/kernel/msm/+/android-msm-hammerhead-3.4-lollipop-mr1.1/Documentation/cpu-freq/governors.txt) is similar to ondemand...

```
The CPUfreq governor "interactive" is designed for latency-sensitive,
interactive workloads. This governor sets the CPU speed depending on
usage, similar to "ondemand" and "conservative" governors.  However,
the governor is more aggressive about scaling the CPU speed up in
response to CPU-intensive activity.
```