blue-systems/plamo-initial-dev_DEPRECATED #95: [home button]: should show the homescreen without staying active

star-buck, 2015-06-20:


notmart, 2015-06-22:
@mgraesslin: can the show desktop effect get automatically disabled when a new window arrives?

mgraesslin, 2015-06-22:
That should be the case - I need to check with Thomas whether all Show Desktop related bug fixes made it into master branch, too.

luebking, 2015-06-22:
Should have made unless my origin/master is wrong ;-)

Notice that new windows will *NOT* break the state *if* they
- "somehow belong to the desktop" (same client or transient for it) or
- are flagged keep above.

mgraesslin, 2015-06-22:
I guess the problem is that ShellClient doesn't break the show desktop state yet

luebking, 2015-06-22:
Indeed - Client breaks it in takeFocus() (rather than on plain showing to (a) keep focus stealing prevention in and (b) allow explicit breaking through window activation in the taskbar and such)

mgraesslin, 2015-06-23:
Fixed with http://commits.kde.org/kwin/7e1e044948ba70bd6521dbc5b782d8bbbe67cd68

Please rebuild kwin

jriddell, 2015-06-23:
building in on-demand...

star-buck, 2015-06-23:
works, just needs visibly remove the blue :)

closed 2015-06-23 by star-buck