blue-systems/plamo-initial-dev_DEPRECATED #96: [volume buttons]: map volume buttons to osd

star-buck, 2015-06-20:


mgraesslin, 2015-06-20:
suggestion: use normal architecture:
* port kglobalaccel to wayland as discussed
* use kmix
* use standard osd functionality of kmix<->plasmashell

mck182, 2015-06-22:
@mgraesslin Sounds like a plan. Although KMix might be unnecessarily heavy. Maybe @apachelogger's new volume applet/its backend would be simpler to use?

Also, has any work on kglobalaccel/wayland started?

mgraesslin, 2015-06-22:
> Also, has any work on kglobalaccel/wayland started?

kglobalaccel is split into plugins. But that's about it.

mgraesslin, 2015-06-26:
some progress on the global accel part: frameworks and kwin reviews are prepared. Once the code gets in, it should magically start to work (assuming the volume applet grabs KMix's keys)

mgraesslin, 2015-07-03:
kglobalaccel, kwin and simplelogin are adjusted. One of the next image should have volume buttons working if the applet has registered global shortcuts.

@apachelogger does your applet grab KMix's global shortcuts?

apachelogger, 2015-07-03:
ENOTIMPLEMENTED

star-buck, 2015-07-08:
On desktop side, installing kmix alongside new volume mixer allows for OSD to work.
On phone adding kmix does not seem to help.
So with kglobalaccel, kwin and simplelogin prepared, is it only registering the shortcuts thats missing?

mgraesslin, 2015-07-08:
starting kmix should register the shortcuts. I can give it a try, to verify whether the shortcuts are installed and get triggered.

What is probably missing is the OSD in plasma-phone.

mgraesslin, 2015-07-08:
I have now tried it on the exopc and there starting kmix makes the normal volume keys work and also an OSD show

apachelogger, 2015-07-08:
it could be that kmix simply bugs out on the phone and doesn't manage to detect any devices. at any rate, I doubt having two apps running just for the OSD is a very reasonable approach anyway :trollface:

mgraesslin, 2015-07-08:
after a rebuild of kglobalaccel and kwin the volume buttons work and Plasma 4's OSD is shown if one manually starts kmix (I needed to use konsole to start it. The launcher didn't work).

star-buck, 2015-07-21:
IMG 135: doesnt work

apachelogger, 2015-07-21:
uh uh. I have been working on plasma-volume and it should now do the proper plasma OSD (via osdservice). assuming the key mapping is correct this actually should now make it work without kmix

http://mobile.kci.pangea.pub:8080/job/vivid_unstable_plasma-volume-control_bin_armhf/2/

star-buck, 2015-07-21:
lets ship updated package in next IMG

star-buck, 2015-07-31:
shipped and initially works

closed 2015-07-31 by star-buck