blue-systems/plamo-initial-dev_DEPRECATED #109: [kci phone]: build ci packages and store on kci repo, then build new IMG 112

star-buck, 2015-06-21:
package repository should be somewhere equivalent of this:
http://kci.pangea.pub/plasma-phone/debian/pool/main/

star-buck, 2015-06-22:
new mobile packages repo home is up here:
http://mobile.kci.pangea.pub/pool/main/

@apachelogger :Lets try spinning a new IMG (112) before today's end.

apachelogger, 2015-06-22:
@star-buck most mobile specific jobs are not built yet there is nothing to build an image from

star-buck, 2015-06-22:
@apachelogger : How many mobile specific are left before we can spin an IMG again?

apachelogger, 2015-06-22:
~20 still need jenkins jobs created and their initial build

star-buck, 2015-06-22:
@apachelogger : how many packages do we have for mobile kci in total?
 Can't you reuse most of the on-demand-ones from dci?

apachelogger, 2015-06-22:
sure. the on demand stuff is from friday though, so the image would be the same as the existing one as far as the phone is concerned

closed 2015-06-23 by star-buck