blue-systems/plamo-initial-dev_DEPRECATED #113: [kwin] some issues with the sliding "windows go below" panel

notmart, 2015-06-23:
the "fullscreen" top sliding panel is marked in kwayland as a panel window and "windows go below": however sometimes some windows seems to still be sized/positioned according to its geometry:
opening a maliit keyboard when the top panel is open (network plasmoid wifi password) will make the keyboard appear moved down, in a way only the first line of keys qwertyuiop is shown.

also, sometimes the launch feedback window (black window with spinner) is also sized to go at the bottom of the sliding panel window.

they are still visible at all because of a problem in sizing that window as well: it's sized excluding the other two actual panels, but moved at 0,0 not considering the top one (hardcoding different sizes in the window size doesn't make any difference, seems that kwin is resizing its window).
so when the empty area gets darker, it's possible to see where the actual window ends.

what this suggests is that the "windows can cover" policy isn't respected everywhere, everytime?

mgraesslin, 2015-06-23:
do you see a pattern for the "sometimes"? I'm currently unable to test as the latest image resulted in a broken plasmashell on startup :-(

notmart, 2015-06-23:
an apt-get update/dist-upgrade should solve it

notmart, 2015-06-23:
the slider window always appear with the wrong size/position
the keyboard called from networkmanager either appears at the wrong position or doesn't appear at all

mgraesslin, 2015-06-24:
is the keyboard marked as fullscreen window, or is it just a window with the same size as the screen? It is important that it's marked as fullscreen to have it not affected by placement and constraints, etc.

mgraesslin, 2015-06-24:
* concerning positioning: the panel doesn't use PlasmaShellSurface's method to position itself, this needs fixing in plasma-mobile. Patch to set to 0/0 is at https://paste.kde.org/pqq5ljzcc/okoatp - please adjust to make the position correctly
* KWin doesn't properly update when the panel behavior changes, have a patch prepared.

In general I find it very difficult to trigger the maliit keyboard on the top panel as it doesn't take focus. I think we need to change that somehow.

mgraesslin, 2015-06-24:
there seems to be also a problem with the call ordering: first panel is set, then panel behavior. This seems to be a problem for the launch feedback window resizing.

mgraesslin, 2015-06-25:
please rebuild kwin

mgraesslin, 2015-06-25:
sizing is fixed with http://commits.kde.org/plasma-mobile/23143ead7992d22b4cf0e6b538dc0628aa111363

Please rebuild plasma-mobile

jriddell, 2015-06-25:
building kwin in mobile kci

closed 2015-06-25 by jriddell

jriddell, 2015-06-25:
building plasma-mobile in mobile kci