blue-systems/plamo-initial-dev_DEPRECATED #115: [Qt] qtquickcontrols not in "touch" mode

notmart, 2015-06-23:
qtquickcontrols change their behavior when touchscreen are present, like hiding scrollbars and show the little copy/paste popups.
however this doesn't happen on the phone.
the code that in QtQuickControls decides this is qquickcontrolssettings.cpp:
bool QQuickControlSettings::hasTouchScreen() const
{
// QTBUG-36007
#if defined(Q_OS_ANDROID)
    return true;
#else
    foreach (const QTouchDevice *dev, QTouchDevice::devices())
        if (dev->type() == QTouchDevice::TouchScreen)
            return true;
    return false;
#endif
}

this means QTouchDevice::devices() returns an empty list, or device of wrong type.

this i guess should come from the wayland platform integration plugin?

mgraesslin, 2015-06-23:
if it comes from the QPA it looks strange as the ifdef would not be needed.

notmart, 2015-06-23:
that comes from qtquickcontrols.

something related i see in the qpa is in QWaylandInputDevice::seat_capabilities
qtwayland/src/client/qwaylandinputdevice.cpp

where a QTouchDevice gets created (they're static trough the app, so i guess that code path should be hit first, then QTouchDevice::devices() from qtquickcontrols should return those created there)

mgraesslin, 2015-06-24:
KWin properly announces that touch is supported. From hwcomposer backend:
waylandServer()->seat()->setHasTouch(true);

notmart, 2015-07-14:
seems the problem is that the controls settings besides signaling to have touch also needs an "ismobile" property, that on linux is actually hardcoded, so we need a fork of it

notmart, 2015-07-14:
that's wit some patches to qtquickcontrols http://imagebin.ca/v/28ahPi51oZI0
unfortunately i don't think is anything mergeable upstream, it would need some patches to be carried out in our package

aleixpol, 2015-07-14:
Why isn't it mergeable? :/ can we move it into our QPT or such?

It looks good by the way.

apachelogger, 2015-09-24:
Whatever needs packaging here?

notmart, 2015-09-28:
it's all fine now as ass the needed patches have been pushed in to Qt
now scrollbars appear and disappear as expected on mobile devices, the inline edit menu appears to work

closed 2015-11-15 by star-buck