blue-systems/plamo-initial-dev_DEPRECATED #117: [phonebook] need a way to import contacts

sebasje, 2015-06-23:
Issue should be know, the phonebook is empty and there's no apparent way to get contacts in there. We'd need to be able to import (and perhaps sync, but import has way higher priority) contacts from online sources such as webdav or google.

mck182, 2015-07-03:
Ok so I have a working prototype code that takes an oauth access token from KAccounts, downloads all Google contacts into vcards and then KPeople reads them and makes them accessible in phone book.

So far it misses some working UI for KAccounts - @aleixpol have you started something on that? - and needs testing the whole setup on the phone (libaccounts+signon+signon-ui+kpeople+kpeoplevcard+kcontacts+kgapi).

aleixpol, 2015-07-03:
How come you decided to turn Google contacts into vcards? Whouldn't it be easier to just have a Google KPoeple backend?

I didn't start because you said there were things much more important than KAccounts UI. I did put some things together though, they can be found in kaccounts-integration/examples.

mck182, 2015-07-03:
They would have to be serialized in one way or another for permanent storage, KGAPI uses KContact::Addressee subclasses for contact objects for which there is a serialization into vcards and there is already a vcard KPeople backend, so it's a win-win situation.

mck182, 2015-07-14:
Update on current progress.

I have successfully imported my google contacts into the phone via kaccounts. In plasma-settings there is now a kded module watching for google account being added into kaccounts and if it detects it, an import is triggered. Phonebook shows these contacts then.

This will now require
 * kgapi frameworks branch packaged (see issue #154)
 * kcontacts package
 * proper QT_IM_MODULE in /etc/environment (see issue #151)
 * kaccounts-integration package installed
 * kaccounts-providers package installed
 * patched signon-ui installed (see issue #153)
 * kpeoplevcard plugin from kde:scratch/apol/kpeoplevcard branch lookupService (issue not yet filed)
 * finished phonebook part of dialer (about 80% there)
 * possibly something more I forgot because of the late hour

Although I do have it running on my phone, I do not deem it usable/done just yet so please wait till I finish & double test everything.

jriddell, 2015-07-15:
* libkgapi is in kci
* kcontacts TODO
* QT_IM_MODULE=maliit >> /etc/environment in mobster
* kaccounts-integration installed
* kaccounts-providers installed
* patched signon-ui TODO
* apt removing signon-kwallet-extension removes kaccounts-integration plasma-settings, what should I do here?
* kpeoplevcard TODO
* phonebook dialer your call (pun intended)
* did you sleep well?

jriddell, 2015-07-16:
* kcontacts is in mobileci http://mobile.kci.pangea.pub/pool/main/k/kcontacts/
* patched signon-ui is also in

jriddell, 2015-07-16:
kpeoplevcard gives me a test failure with latest kpeople
********* Start testing of KPeopleVCardTest *********
Config: Using QtTest library 5.4.2, Qt 5.4.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 4.9.2)
FAIL!  : KPeopleVCardTest::initTestCase() 'm_vcardsDir.exists()' returned FALSE. ()
   Loc: [/home/jr/src/kpeoplevcard/kdegit/kpeoplevcard/autotests/kpeoplevcardtest.cpp(50)]
PASS   : KPeopleVCardTest::cleanupTestCase()
Totals: 1 passed, 1 failed, 0 skipped, 0 blacklisted
********* Finished testing of KPeopleVCardTest *********

aleixpol, 2015-07-16:
Fixed

jriddell, 2015-07-16:
* kpeoplevcard plugin now in on-demand @apachelogger should add it to mobile ci now

now waiting on @mck182's points
* finished phonebook part of dialer (about 80% there)
* possibly something more I forgot because of the late hour

mck182, 2015-07-16:
Right, the phonebook is in.

There are basically two things left - sometimes kgapi returns 0 contacts and no error (talking to dvratil about it now) and then signon-kwallet-extension crashes. Easiest would be to not have it as a dependency on kaccounts-integration, but I'll try to investigate what's wrong with it.

aleixpol, 2015-07-16:
kpeoplevcard was moved into playground, so please drop the scratch/apol/ :)

jriddell, 2015-07-17:
is there packaging to be done for this phonebook?

mck182, 2015-07-17:
No. It's all part of plasma-phone-components, so already there.

star-buck, 2015-07-21:
IMG 135: phonebook is not doing anything? if i press on "Settings" nothing happens, but the button stays active even if selecting another options from the top...

mck182, 2015-07-21:
The phonebook is part of the dialer, so you'd need to run the dialer and select contacts there.

If you see a settings button, you're probably running the phonebook component tester which shouldn't be used.

star-buck, 2015-07-21:
And where are the contacts in the dialer under that tab? How can I add
contacts there?
The phonebook is the phonebook shipped as in IMG135. That should not be
installed, wrong version or what do you mean with "that shouldnt be used?"

mck182, 2015-07-21:
Let me flash that image and have a look.

mck182, 2015-07-21:
Ok so the Phonebook app that is there is just the components tester, not the app. I'll make sure this is not installed.

The phonebook itself is currently in Phone -> Contacts. I'll see if a button like "Import contacts" can be added there.

star-buck, 2015-07-21:
Maybe its better to seperate that into phonebook (what others call
contacts)?

The workflow should be:
-add an account, where it has possible contacts as resource (via kaccounts
kcm under settings)
- import contacts into phonebook from available accounts (phonebook should
sync back contacts to external kaccount if edited, removed or added a new
one manually?)
-see contacts in dialer which have at least one valid phone number (thats
probably why most seperate apps are called contacts and not phonebook, so
contacts/adressbook should be tied to kpeople, where a person can have
multiple contacts from various accounts?)

I thought that approx. logic was already there in kaccounts/kpeople on
desktop, so only the dialer needed to filter valid contacts with phone
numbers?

mck182, 2015-07-21:
That is moreless how it works, the current workflow on the phone is as follows:
 * add account in kaccounts in settings
 * contacts from google start importing
 * dialer now lists all contacts with a phone number, allowing you to call them directly

I guess a standalone Contacts/Addressbook with all your contacts could also be added.

star-buck, 2015-07-21:
Okay, good.
So how can I add my google account in IMG 135 to get those contacts showing
up in the dialer?

mck182, 2015-07-21:
It would appear that plasma-settings package is of an older date (0.0.1ubuntu4+git20150715.1133+15.04-0) and is not upgradeable on the phone (complaining about breaking other packages).

It is in latest build of plasma-settings (0.0.1ubuntu4+git20150717.0959+15.04-0) though.

mck182, 2015-07-21:
Actually scratch that, you can update it with "apt-get install plasma-settings". It complains if you do upgrade && dist-upgrade which will remove it first and then doing install complains.

mck182, 2015-07-21:
Furthermore, issue #153 needs to be resolved first; signon-ui fails to build on arm for unknown reasons (the compiler seems to crash?) and so signon-ui crashes when adding new account.

star-buck, 2015-09-24:
is this fixed by now?

mck182, 2015-09-24:
For some reason the new built package of signon-ui does not work on the phone. Investigating.

bhush9, 2016-01-20:
@mck182  since the #153 is now solved, I think this works now? should I mark it as ready for QA?

mck182, 2016-01-26:
Yes, with all packages upgraded I managed to import my Google contacts just fine.

star-buck, 2016-02-13:
@mck182 :
is this working with latest IMG20/xenial?

mck182, 2016-02-19:
No; there was a major architectural change which is partly upstream. 

1 out of 3 changes upstream is merged, the other two should be done quickly on Monday or whenever upstream reviews it. After that's done, I'll push min version requirement update to KAccounts, the image will then require master or latest versions of libaccounts-glib and libaccounts-qt as well as kaccounts-integration. Then it will work again.

I'm coordinating with bshah on this; it should be ready for next week's image.

bhush9, 2016-03-01:
importing contacts works fine in IMG27