blue-systems/plamo-initial-dev_DEPRECATED #118: [kaccounts]: make useable as kcm under system settings

star-buck, 2015-06-23:


aleixpol, 2015-06-23:
See kaccounts-integration/example/accounts.qml, should be rather easy to turn into a kcm.

Makes use of upstream Accounts SSO QML Module:
https://gitlab.com/accounts-sso/accounts-qml-module

star-buck, 2015-07-07:
has this been worked on or already packaged?

mck182, 2015-07-07:
I'm working on it, not yet done.

mck182, 2015-07-10:
I've pushed a basic working version of KAccounts into plasma-settings, it requires newest libkaccounts+qml-module-org-kde-kaccounts packages. Adding Google account works fine on desktop however I wasn't able to test on the phone as the keyboard wouldn't come up to enter my credentials. So far the Google account doesn't do anything yet, but adding the contacts import is fairly trivial now as I have the code written and working, so I'll work on that next.

aleixpol, 2015-07-10:
Shouldn't it set up the Telepathy account?

mck182, 2015-07-11:
Eventually yes. The current aim is to get contacts import/sync done from Google so that you can have contacts in your phone book.

sebasje, 2015-07-13:
I see an Accounts module now in settings, but it fails to load.

mck182, 2015-07-13:
It requires newest libkaccounts+qml-module-org-kde-kaccounts packages. Please make sure you have them installed.

If you do, please run it from konsole and post the error output.

sebasje, 2015-07-13:
I have updated everything just before running. Are all the packages built and will be installed from apt-get dist-upgrade?

Packages are still rebuilding, so perhaps it hasn't pulled in the right dependencies yet. I'll check again later on.

mck182, 2015-07-13:
No, everything is already built (I made sure on Friday), but you need to explicitly install qml-module-org-kde-kaccounts, it's a new package which wasn't there before (and it's not a dep yet afaik).

sebasje, 2015-07-13:
Ah, cool. Now it loads, but only offers Ubuntu One accounts to add. After installing account-plugin-google and libaccount-plugin-google, I also get Google as option, but nothing happens when I tap on it? (Same as for the Ubuntu One, the button doesn't seem to do anything)

Do I need an additional package for Google accounts, for example?

Can you file a ticket to add qml-module-org-kde-kaccounts to the deps of plasma-settings?

aleixpol, 2015-07-13:
@sebasje Maybe it also needs kaccounts-providers?

mck182, 2015-07-13:
Ah you perhaps also require signon-ui. Can you check if you have that? kaccounts-providers is not yet packaged I think, but you should be just fine with account-plugin-google for now (it has the same file).

sebasje, 2015-07-13:
Installing signon-ui brings up a webview which wants me to sign in to Google, but I can't fill in anything as the virtual keyboard doesn't come up.

Guys, can you have a look into this? Testing something on the device itself is part of making it work, and the addition of a few dependencies of this package could be done by yourselves much much easier...

mck182, 2015-07-13:
Well I have already written in the first comment 3 days ago "I wasn't able to test on the phone as the keyboard wouldn't come up to enter my credentials". I understand Marco & Martin G are looking into the keyboard issue.

Signon-ui should have been a dependency, yes, I already had it installed so I thought it's there be default as the ubuntutouch also needs it (and I did mention you require qml-module-org-kde-kaccounts to be installed).

I'll file new issue for the dependencies.

sebasje, 2015-07-13:
mck182: thanks!

mck182, 2015-07-13:
Actually there's one more thing - the signon-ui package distributed by ubuntu has a bug in it; it assumes unity running "if (platform != xcb)" so the phone with "platform == wayland" hits it and signon-ui thinks it's running in unity and so requires ubuntu-browser or somesuch to be present.

I've sent the patch to Alberto (maintainer) and it's awaiting merging. In the meantime, we'll likely require a package off a bazaar branch to not have to deal with tons of unity weirdness. I'll do a clean image test once this wild "unexpected EOF" error be gone when trying to flash.

For tracking purposes:
https://bugs.launchpad.net/ubuntu/+source/signon-ui/+bug/1473422

star-buck, 2015-07-21:
can this be packaged and shipped?

star-buck, 2015-07-31:
present in Settings

closed 2015-07-31 by star-buck