blue-systems/plamo-initial-dev_DEPRECATED #119: Dialer: crash as soon tries to dial

notmart, 2015-06-23:
unfortunately don't have much debug symbols on the phone, that's what gdb says:

Program received signal SIGSEGV, Segmentation fault.
0xb65eb6a8 in Tp::Connection::contactManager() const ()
   from /usr/lib/arm-linux-gnueabihf/libtelepathy-qt5.so.0
(gdb) bt
#0  0xb65eb6a8 in Tp::Connection::contactManager() const ()
   from /usr/lib/arm-linux-gnueabihf/libtelepathy-qt5.so.0
#1  0x00019d36 in DialerUtils::dial(QString const&) ()
#2  0x0001a30c in DialerUtils::qt_metacall(QMetaObject::Call, int, void**) ()
#3  0xb63a1040 in ?? () from /usr/lib/arm-linux-gnueabihf/libQt5Qml.so.5
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

star-buck, 2015-06-24:
Reference ticket is here:
https://github.com/plasma-mobile/mobster/issues/27

mck182, 2015-06-24:
Interesting, I can make a successful phone call when launched from terminal with the env from /usr/bin/plasma-phone but it does not work when executed from the launcher.

Investigating.

mck182, 2015-06-24:
Ok I know what's up. It's AppArmor screwing with the Mission Control plugin. The plugin runs some commands to find out how many modems there are present on the device. This is getting denied by AppArmor. 

If anyone understands AppArmor, your help would be most welcomed on adjusting the telepathy profile.

As a workaround, FORCE_RIL_NUM_MODEMS=1 can be set to not go via the commands. I've put it to plasma-phone script but apparently that is too late as mission-control-5 is started before that. So it needs to go to /etc/environment, then phone calls work as expected.

When #121 is solved, this env var can just be added there. Otherwise the AppArmor profile needs to be fixed but I cannot make any sense of it.

notmart, 2015-06-24:
yeah, setting that env var in /etc/environment seems to solve the problem.
we should also move most of our env stuff in there i guess.
how to generate an image with changes to that file?

mck182, 2015-06-24:
That should be done by now

[16:45:42] <mck182> Riddell: can you also add FORCE_RIL_NUM_MODEMS=1 to the plasma-phone-settings?
[16:45:48] <mck182> to /etc/environment
[17:06:57] <Riddell> mck182: ok
[17:07:49] <Riddell> mck182: pushed to git, waiting on mobile ci reappearing to update packages

notmart, 2015-06-24:
now most of the time when it tries to ring it does this
ringtone gets played correctly, i guess by voicecall, but the phonedialer can crash with this error.

also, when it doesn't crash, dialerUtils.callState doesn't seem to get updated, so the loader with CallPage.qml never loads it never showing the answer/decline ui

#0  0xb67fec10 in KNotification::sendEvent() ()
   from /usr/lib/arm-linux-gnueabihf/libKF5Notifications.so.5
#1  0x0001c7ea in CallManager::onCallStateChanged(Tp::CallState) ()
#2  0x0001cf00 in CallManager::CallManager(Tp::SharedPtr<Tp::CallChannel> const&, DialerUtils*, QObject*) ()
#3  0x0001adea in CallHandler::handleChannels(Tp::MethodInvocationContextPtr<Tp::MethodInvocationContextTypes::Nil, Tp::MethodInvocationContextTypes::Nil, Tp::MethodInvocationContextTypes::Nil, Tp::MethodInvoca

star-buck, 2015-06-24:
triggered rebuild of plasma-phone-settings

star-buck, 2015-06-26:
still crashing as of IMG 120

mck182, 2015-06-26:
The crash should be fixed now, I can accept incoming calls with latest plasma-phone-components package.

There is still a problem with the dbus activation (incoming call when the app is not running), for some reason the app registers on dbus but it never replies to the dbus activation call, making things time out and not work properly or sometimes make the app start second time. I'm continuing to investigate this.

mck182, 2015-06-30:
Ok the issue seems to actually be caused by package telephony-service, which is Ubuntu's dialer app. This installs /usr/bin/ofono-setup which is a) unnecessary and b) starts mission-control sooner than plasma does, resulting in two different mission-control-5 processes running, however that one is started from different env so it does not have the env variable needed, breaking things.

Doing sudo apt-get remove telephony-service makes things work on freshly flashed img 122. If someone could verify that'd be awesome.

mck182, 2015-07-03:
The crash is solved as of img 123.

star-buck, 2015-07-07:
closing this then.

star-buck, 2015-07-07:
again crashes here as of IMG 129

star-buck, 2015-07-07:
after a reboot, it seemed to not crah anymore... so it needs root write access to function?

mck182, 2015-07-07:
It seems to be NetworkManager related. When there's no wifi, MissionControl will not connect (and not even attempt to).

Can you confirm that if you have no wifi set up (nmcli connection delete $connection), reboot, it doesn't work, connect to wifi and it starts working?

star-buck, 2015-07-08:
seems indeed so, that is a (logical) bug then?

mck182, 2015-07-08:
MissionControl has a built-in NetworkManager plugin so if you request your account to go online (ie. register to GSM network), it first checks if it has a connection via NM. If it doesn't, it just waits till it gets a network connection and only then actually connects. And without the account having a connection one is unable to make calls.

I'll see if the ofono MC account can be taken out of this NM conditioning.

mck182, 2015-07-08:
Ok so I've found two bugs, one in mission-control and one in telepathy-ofono. I'm not sure how fast/if at all this can get to upstream however, so I'll talk to our packagers tomorrow if we can packager our patches on top and ship our own packages.

mck182, 2015-07-08:
For reference, bug with patch filed upstream against telepathy-mission-control: https://bugs.freedesktop.org/show_bug.cgi?id=91272

Telepathy-ofono maintainer has been sent a patch as well.

mck182, 2015-07-15:
Telepathy-ofono patch at https://code.launchpad.net/~tiagosh/telepathy-ofono/fix-mc-plugin-property-set/+merge/264198

star-buck, 2015-07-31:
@mck182: if fixed please close.

mck182, 2015-07-31:
Works fine on the latest image.

closed 2015-07-31 by mck182