blue-systems/plamo-initial-dev_DEPRECATED #120: [KWayland]: mapping of touch-to-mouse events X11-clients

star-buck, 2015-06-24:
code to make it work should be already package, but not working?:
http://mobile.kci.pangea.pub/pool/main/k/kwayland/

star-buck, 2015-06-24:
@jriddell :
can you check the commit from notmart is actually packaged and in the latest IMG?

notmart, 2015-06-24:
the commit id is cde26bc71c7148381e82799c17509966b03a93fb of kwayland repo

mgraesslin, 2015-06-24:
possible fix in kwin: http://commits.kde.org/kwin/b8ae823b50a84e8c2355ed8dfb75cf18bd60a056

Please rebuild kwin

jriddell, 2015-06-24:
waiting on mobile ci to re-appear

star-buck, 2015-06-24:
triggered rebuild of kwin

star-buck, 2015-07-07:
IMG 129: typing "gnome-chess" in konsole doesnt even start the game anymore (worked before)

star-buck, 2015-07-07:
instead gnome-chess freezes the screen

notmart, 2015-07-16:
right now x11 apps make kwin crash, however with the next packaging of the breeze window deco (weird but has to do about it) it should work: successfully managed to move pawns in gnome-chess ;)

star-buck, 2015-07-31:
works, but still shaky: 
first touch to select a figure works consitently, but moving to a new field only works ~20% of the time...

as a clue for the first touch event to pick a figure, one can reproducably get this behaviour:
select a figure, then deselect it again by pressing again on the same figure.
Then selecting a new figure elsewhere on the baord activates precisely the former figure again, as if the third touch event is still executed on the former piece area.

@notmart : any idea how to improve this to work more reliable?

notmart, 2015-09-28:
i changed it a bit faking a move before the touch. It seems to work better on the x86 system, let's see how will be on the phone

star-buck, 2015-11-15:
this worked before, but with IMG11 seems to have stopped working.

mgraesslin, 2015-11-15:
actually there is a better way now: we "just" need to ship Xwayland 1.18 (which got released last week). It now has proper support for touch events built in. I want to open a dedicated issue for that anyway.

mgraesslin, 2015-11-15:
and created #217 for that.

bhush9, 2016-01-20:
now xwayland 1.18 is shipped, marking as Read for QA.

star-buck, 2016-01-20:
works.

closed 2016-01-20 by star-buck