blue-systems/plamo-initial-dev_DEPRECATED #121: [packaging]: set Path env default to include: /usr/games/

star-buck, 2015-06-24:


jriddell, 2015-06-24:
my desktop has:
>cat /etc/environment 
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games"

but I've no idea what makes that file

mgraesslin, 2015-06-24:
I just opened konsole on the phone:
echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

jriddell, 2015-06-24:
I believe you!  There's something which overwrites /etc/environment in touch so I guess I'll just add it to our setup package to fix it

jriddell, 2015-06-24:
added to plasma-phone-settings, will be updated when mobile ci reappears

jriddell, 2015-06-24:
building...

star-buck, 2015-06-25:
IMG 120: still not working with latest plsma-phone-settings, $PATH:
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

apachelogger, 2015-06-26:
@jriddell PATH comes from /etc/login.defs initially, so unless something manually mangles the PATH after login (e.g. sets it explicitly to the limited version) this is either a problem with simplelogin/pam not loading login.defs correctly or login.defs being wrong.

I feel like we had the same problem with sddm a while ago and it turned out to be the PAMs IIRC

jriddell, 2015-06-30:
just added to /etc/environment for now

closed 2015-06-30 by jriddell