blue-systems/plamo-initial-dev_DEPRECATED #124: [app] Provide a screenshot app

mgraesslin, 2015-06-24:
Based on a small discussion yesterday on IRC I made it possible to use KWin's screenshot dbus interface with OpenGL ES 3.0. This is available in latest kwin master.

With that it's possible to create a fullscreen screenshot in the following way:
qdbus org.kde.KWin /Screenshot org.kde.kwin.Screenshot.screenshotFullscreen

The call returns the path where the file was created, e.g.:
/tmp/kwin_screenshot_T12709.png

A possible app could be just a desktop file which performs the dbus call or a with a small qml interface to set a screenshot delay.

bhush9, 2015-08-04:
I've played little bit with it today

[image]

Here is initial version, at moment of commenting it is very basic app

bhush9, 2015-09-26:
Packagers please package it from : https://github.com/plasma-apps/screenshotapp

(I will eventually move it to kde git)

star-buck, 2015-11-16:
shipped and kinda works

closed 2015-11-16 by star-buck