blue-systems/plamo-initial-dev_DEPRECATED #127: [click]: run click based applications so they come up

star-buck, 2015-06-24:
make apparmor support wayland (Error log):

wl_display@1: error 1: invalid arguments for android_wlegl_handle@38.add_fd
The Wayland connection experienced a fatal error (Invalid argument)
Wayland display got fatal error 22: Invalid argument
Additionally, errno was set to 22: Invalid argument
The display is now unusable, aborting.

star-buck, 2015-06-24:
Been playing a bit further, my conclusion is that click already works, but we need to be able to integrate applications written in click.

A first step will be to include ```/userdata/user-data/phablet/.local/share/applications/``` in the launcher list. This won't work on all applications though (e.g. the gallery app) because to execute the applications they use ```aa-exec-click``` that checks whether it's running Mir. This will make the desktop files not to work sometimes. Nevertheless, some applications such as the one described above will just work.

So thing is it's already in the look-up path, even though it doesn't work at the moment, because it's using aa-exec-click. The only reason why it's not working is because there's a hook that modifies the desktop file to ensure it's going to be executed under apparmor.

The error message: ```apparmor="DENIED" operation="connect" profile="net.launchpad.click-webapps.xkcd_xkcd_5" name="/run/user/32011/wayland-0" pid=5290 comm="QtWayland" requested_mask="rw" denied_mask="rw" fsuid=32011 ouid=32011```

Here it gets messy:
* We can disable the apparmor hook. Hooks can be edited with ```find /usr -name "*\.hook"``` and disabled, theoretically with ```click hook remove ubuntu-app-launch-desktop run-system```. Only that in practice it doesn't work, or I haven't been able to.
* We can fix apparmor so that applications get to be launcher properly. Now this sounds very reasonable, since we'll be leveraging an interesting technology anyway, in principle. I've spent about 3h trying to make it work, but didn't manage. What we need is to make sure the application can connect to ```/run/user/<pid>/wayland-0```, should be easy but I wasn't able to alter the behavior work.

The generated execution line is ```aa-exec-click -p net.launchpad.click-webapps.xkcd_xkcd_5 -- webapp-container --webappModelSearchPath . --enable-back-forward --homepage http://m.xkcd.com/```

We did some progress, with the help of @apachelogger. Saving here, just in case I might flash my phone:
```
phablet@ubuntu-phablet:~$ cat /etc/apparmor.d/abstractions/wayland
owner /run/user/*/wayland-0 rw,
owner /run/user/*/weston-shared-* rw,
owner /tmp/wayland-shm-* rw,
```
Still, will need to keep whitelisting stuff.

star-buck, 2015-06-24:
To run click-based applications, like preinstalled calculator:
* Add the file specified in the comment before.
* ```echo "#include <abstractions/wayland>" >> /etc/apparmor.d/abstractions/base```
* prevent aa-exec-click from closing if there's some .X stuff.
```
phablet@ubuntu-phablet:/usr/share/click/preinstalled/com.ubuntu.calculator/2.0.182$ diff /usr/bin/aa-exec-click.original /usr/bin/aa-exec-click
54c54
<         exit 1
---
> #        exit 1
```

notmart, 2015-06-24:
the new file is correctly here, but it still needs some files installed by upstream packages to be patched: how that should be done?

the packages are:
* apparmor: file /etc/apparmor.d/abstractions/base
* click-apparmor: file /usr/bin/aa-exec-click

we need to either have our patched build of those packages or to patch those 2 files at startup (that looks much more error prone)

star-buck, 2015-06-25:
@jriddell : please package the above files

jriddell, 2015-06-25:
my changes are in git but don't seem to be picked up by mobile kci which is using an older version of plasma-phone-settings-packaging repository

jriddell, 2015-06-25:
ah hah, I just didn't heed harald's warnings about using kubuntu_unstable branches

jriddell, 2015-06-25:
plasma-phone-settings 4+git20150625.1605+15.04 should have it.  if you are installing on an existing system then apt remove --purge plasma-phone-settings; apt install plasma-phone-settings  as it will only do the dpkg-divert on a new install

closed 2015-06-25 by jriddell