blue-systems/plamo-initial-dev_DEPRECATED #128: [SDK] Dependencies broken

aleixpol, 2015-06-25:
After updating the sources.list I just triggered the setup.sh on my system again, to get a new setup, then I found it wasn't working.

That's the one we have now, what should we have?
```
deb [arch=armhf] http://ports.ubuntu.com/ vivid main universe restricted
deb-src [arch=armhf]  http://ports.ubuntu.com/ vivid main universe restricted

deb [arch=armhf] http://ports.ubuntu.com/ vivid-updates main universe restricted
deb-src [arch=armhf] http://ports.ubuntu.com/ vivid-updates main universe restricted

deb [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid main universe restricted
deb-src [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid main universe restricted

deb [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid-updates main universe restricted
deb-src [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid-updates main universe restricted

deb [arch=armhf] http://mobile.kci.pangea.pub vivid main
deb-src [arch=armhf] http://mobile.kci.pangea.pub vivid main
```

apachelogger, 2015-06-25:
What's the error?

aleixpol, 2015-06-25:
```
root@oliver:~# apt-get build-dep -a armhf kwin
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages have unmet dependencies:
 kdoctools-dev:armhf : Depends: kdoctools5:armhf (= 5.10.0+git20150622.1658+15.04-0) but it is not going to be installed
 kio-dev:armhf : Depends: libkf5bookmarks-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5completion-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5jobwidgets-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5service-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5xmlgui-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5config-dev : Depends: libkf5configcore5 (= 5.9.0-0ubuntu1) but it is not going to be installed
                    Depends: libkf5configgui5 (= 5.9.0-0ubuntu1) but it is not going to be installed
                    Depends: libkf5config-bin (= 5.9.0-0ubuntu1) but it is not going to be installed
                    Depends: qtbase5-dev
 libkf5coreaddons-dev : Depends: libkf5coreaddons5 (= 5.9.0-0ubuntu1) but it is not going to be installed
                        Depends: qtbase5-dev
 libkf5crash-dev:armhf : Depends: libkf5coreaddons-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5declarative-dev:armhf : Depends: libkf5package-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5kcmutils-dev:armhf : Depends: libkf5configwidgets-dev:armhf (>= 5.10.0) but it is not going to be installed
                            Depends: libkf5iconthemes-dev:armhf (>= 5.10.0) but it is not going to be installed
                            Depends: libkf5service-dev:armhf (>= 5.10.0) but it is not going to be installed
                            Depends: libkf5xmlgui-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5newstuff-dev:armhf : Depends: libkf5xmlgui-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5notifications-dev:armhf : Depends: libkf5config-dev:armhf (>= 5.10.0) but it is not going to be installed
                                 Depends: libkf5coreaddons-dev:armhf (>= 5.10.0) but it is not going to be installed
 plasma-framework-dev:armhf : Depends: libkf5package-dev:armhf (>= 5.10.0) but it is not going to be installed
                              Depends: libkf5service-dev:armhf (>= 5.10.0) but it is not going to be installed
E: Build-dependencies for kwin could not be satisfied.
```

apachelogger, 2015-06-25:
libkf5coreaddons-dev : Depends: libkf5coreaddons5 (= 5.9.0-0ubuntu1) but it is not going to be installed

that doesn't seem right. maybe simply remove the architecture tag from the mobile.kci entries.

aleixpol, 2015-06-25:
```
root@oliver:~# cat /etc/apt/sources.list
deb [arch=armhf] http://ports.ubuntu.com/ vivid main universe restricted
deb-src [arch=armhf]  http://ports.ubuntu.com/ vivid main universe restricted

deb [arch=armhf] http://ports.ubuntu.com/ vivid-updates main universe restricted
deb-src [arch=armhf] http://ports.ubuntu.com/ vivid-updates main universe restricted

deb [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid main universe restricted
deb-src [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid main universe restricted

deb [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid-updates main universe restricted
deb-src [arch=i386,amd64] http://archive.ubuntu.com/ubuntu vivid-updates main universe restricted

deb http://mobile.kci.pangea.pub vivid main
deb-src http://mobile.kci.pangea.pub vivid main

root@oliver:~# sudo apt-get update
Ign http://archive.ubuntu.com vivid InRelease
Ign http://archive.ubuntu.com vivid-updates InRelease
Ign http://ports.ubuntu.com vivid InRelease
Hit http://archive.ubuntu.com vivid Release.gpg
Get:1 http://archive.ubuntu.com vivid-updates Release.gpg [933 B]
Get:2 http://mobile.kci.pangea.pub vivid InRelease [3183 B]
Ign http://mobile.kci.pangea.pub vivid InRelease
Ign http://ports.ubuntu.com vivid-updates InRelease
Hit http://archive.ubuntu.com vivid Release
Get:3 http://archive.ubuntu.com vivid-updates Release [63.5 kB]
Hit http://ports.ubuntu.com vivid Release.gpg
Ign http://mobile.kci.pangea.pub vivid/main Sources/DiffIndex
Hit http://ports.ubuntu.com vivid-updates Release.gpg
Ign http://mobile.kci.pangea.pub vivid/main amd64 Packages/DiffIndex
Hit http://ports.ubuntu.com vivid Release
Hit http://archive.ubuntu.com vivid/main Sources
Ign http://mobile.kci.pangea.pub vivid/main armhf Packages/DiffIndex
Hit http://ports.ubuntu.com vivid-updates Release
Hit http://archive.ubuntu.com vivid/universe Sources
Hit http://archive.ubuntu.com vivid/restricted Sources
Hit http://ports.ubuntu.com vivid/main Sources
Hit http://archive.ubuntu.com vivid/main amd64 Packages
Hit http://archive.ubuntu.com vivid/universe amd64 Packages
Hit http://ports.ubuntu.com vivid/universe Sources
Hit http://archive.ubuntu.com vivid/restricted amd64 Packages
Hit http://ports.ubuntu.com vivid/restricted Sources
Hit http://archive.ubuntu.com vivid/main i386 Packages
Hit http://archive.ubuntu.com vivid/universe i386 Packages
Hit http://ports.ubuntu.com vivid/main armhf Packages
Hit http://archive.ubuntu.com vivid/restricted i386 Packages
Hit http://archive.ubuntu.com vivid/main Translation-en
Hit http://ports.ubuntu.com vivid/universe armhf Packages
Hit http://archive.ubuntu.com vivid/restricted Translation-en
Hit http://ports.ubuntu.com vivid/restricted armhf Packages
Hit http://archive.ubuntu.com vivid/universe Translation-en
Get:4 http://archive.ubuntu.com vivid-updates/main Sources [51.4 kB]
Hit http://ports.ubuntu.com vivid/main Translation-en
Hit http://ports.ubuntu.com vivid/restricted Translation-en
Get:5 http://archive.ubuntu.com vivid-updates/universe Sources [22.0 kB]
Hit http://ports.ubuntu.com vivid/universe Translation-en
Get:6 http://archive.ubuntu.com vivid-updates/restricted Sources [28 B]
Get:7 http://archive.ubuntu.com vivid-updates/main amd64 Packages [119 kB]
Hit http://ports.ubuntu.com vivid-updates/main Sources
Hit http://ports.ubuntu.com vivid-updates/universe Sources
Hit http://ports.ubuntu.com vivid-updates/restricted Sources
Get:8 http://archive.ubuntu.com vivid-updates/universe amd64 Packages [68.4 kB]
Hit http://ports.ubuntu.com vivid-updates/main armhf Packages
Hit http://ports.ubuntu.com vivid-updates/universe armhf Packages
Get:9 http://archive.ubuntu.com vivid-updates/restricted amd64 Packages [28 B]
Hit http://ports.ubuntu.com vivid-updates/restricted armhf Packages
Get:10 http://archive.ubuntu.com vivid-updates/main i386 Packages [118 kB]
Hit http://ports.ubuntu.com vivid-updates/main Translation-en
[... 19 lines cut ...]
 kdoctools-dev:armhf : Depends: kdoctools5:armhf (= 5.10.0+git20150622.1658+15.04-0) but it is not going to be installed
 kio-dev:armhf : Depends: libkf5bookmarks-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5completion-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5jobwidgets-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5service-dev:armhf (>= 5.10.0) but it is not going to be installed
                 Depends: libkf5xmlgui-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5config-dev : Depends: libkf5configcore5 (= 5.9.0-0ubuntu1) but it is not going to be installed
                    Depends: libkf5configgui5 (= 5.9.0-0ubuntu1) but it is not going to be installed
                    Depends: libkf5config-bin (= 5.9.0-0ubuntu1) but it is not going to be installed
                    Depends: qtbase5-dev
 libkf5coreaddons-dev : Depends: libkf5coreaddons5 (= 5.9.0-0ubuntu1) but it is not going to be installed
                        Depends: qtbase5-dev
 libkf5crash-dev:armhf : Depends: libkf5coreaddons-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5declarative-dev:armhf : Depends: libkf5package-dev:armhf (>= 5.10.0) but it is not going to be installed
 libkf5kcmutils-dev:armhf : Depends: libkf5configwidgets-dev:armhf (>= 5.10.0) but it is not going to be installed
[... cut ...]
 libkf5notifications-dev:armhf : Depends: libkf5config-dev:armhf (>= 5.10.0) but it is not going to be installed
                                 Depends: libkf5coreaddons-dev:armhf (>= 5.10.0) but it is not going to be installed
 plasma-framework-dev:armhf : Depends: libkf5package-dev:armhf (>= 5.10.0) but it is not going to be installed
                              Depends: libkf5service-dev:armhf (>= 5.10.0) but it is not going to be installed
E: Build-dependencies for kwin could not be satisfied.
```

apachelogger, 2015-06-25:
ah yes. the problem is that we do not do amd64 builds via MCI which sort of kind of entirely breaks the notion of xcompiling as now there are no amd64 versions of version 123 so apt falls back to the ubuntu archive version 122 which is broken for xcompile and probably wrong from a version consistency point of view anyway.

bigish issue blocked on https://github.com/plasma-mobile/mobster/issues/116 once we have credentials the CI tech needs to be expanded -.-

I don't see this getting resolved before weekend.

aleixpol, 2015-06-25:
Ok, will check again when I'm back in Europe!

aleixpol, 2015-06-30:
Is this issue redundant to https://github.com/plasma-mobile/mobster/issues/133?

apachelogger, 2015-06-30:
kind of, #133 as I understand it is more about getting images on !arm which in turn would happen to fix this as well so they could be merged if that makes you happy :recycle:

aleixpol, 2015-06-30:
I don't know, I haven't slept much.
Anyway, #116 was solved, what's the status?

apachelogger, 2015-06-30:
WIP :construction_worker:
mostly blocked on me figuring out a job architecture reshuffle to accomodate multiple build architectures

apachelogger, 2015-07-06:
:palm_tree: should be fixed now (for amd64 anyway) since everything has mobile.ci amd64 builds :palm_tree:

instead of the apt sources url
http://ppa.launchpad.net/plasma-phone/ci/ubuntu
the sdk now needs
http://mobile.kci.pangea.pub

aleixpol, 2015-07-06:
I still get the same problem...

```
Reading package lists...
Building dependency tree...
Reading state information...
The following packages have unmet dependencies:
 kdoctools-dev:armhf : Depends: kdoctools5:armhf (= 5.11.0+git20150702.0927+15.04-0) but it is not going to be installed
 kinit-dev:armhf : Depends: kinit:armhf (= 5.11.0+git20150702.1035+15.04-0) but it is not going to be installed
 kio-dev:armhf : Depends: kio:armhf (= 5.11.0+git20150702.1154+15.04-0) but it is not going to be installed
                 Depends: libkf5bookmarks-dev:armhf (>= 5.11.0~) but it is not going to be installed
                 Depends: libkf5completion-dev:armhf (>= 5.11.0~) but it is not going to be installed
                 Depends: libkf5itemviews-dev:armhf (>= 5.11.0~) but it is not going to be installed
                 Depends: libkf5jobwidgets-dev:armhf (>= 5.11.0~) but it is not going to be installed
                 Depends: libkf5kiocore5:armhf (= 5.11.0+git20150702.1154+15.04-0) but it is not going to be installed
                 Depends: libkf5kiofilewidgets5:armhf (= 5.11.0+git20150702.1154+15.04-0) but it is not going to be installed
                 Depends: libkf5kiontlm5:armhf (= 5.11.0+git20150702.1154+15.04-0) but it is not going to be installed
                 Depends: libkf5kiowidgets5:armhf (= 5.11.0+git20150702.1154+15.04-0) but it is not going to be installed
                 Depends: libkf5service-dev:armhf (>= 5.11.0~) but it is not going to be installed
                 Depends: libkf5solid-dev:armhf (>= 5.11.0~) but it is not going to be installed
                 Depends: libkf5xmlgui-dev:armhf (>= 5.11.0~) but it is not going to be installed
 libkf5crash-dev:armhf : Depends: libkf5coreaddons-dev:armhf (>= 5.11.0~) but it is not going to be installed
                         Depends: libkf5crash5:armhf (= 5.11.0+git20150702.0947+15.04-0) but it is not going to be installed
                         Depends: libkf5windowsystem-dev:armhf (>= 5.11.0~) but it is not going to be installed
 libkf5declarative-dev:armhf : Depends: libkf5declarative5:armhf (= 5.11.0+git20150702.1035+15.04-0) but it is not going to be installed
                               Depends: libkf5package-dev:armhf (>= 5.11.0~) but it is not going to be installed
                               Depends: libkf5quickaddons5:armhf (= 5.11.0+git20150702.1035+15.04-0) but it is not going to be installed
 libkf5kcmutils-dev:armhf : Depends: libkf5configwidgets-dev:armhf (>= 5.11.0~) but it is not going to be installed
                            Depends: libkf5iconthemes-dev:armhf (>= 5.11.0~) but it is not going to be installed
                            Depends: libkf5itemviews-dev:armhf (>= 5.11.0~) but it is not going to be installed
                            Depends: libkf5kcmutils5:armhf (= 5.11.0+git20150702.1259+15.04-0) but it is not going to be installed
                            Depends: libkf5service-dev:armhf (>= 5.11.0~) but it is not going to be installed
                            Depends: libkf5xmlgui-dev:armhf (>= 5.11.0~) but it is not going to be installed
 libkf5newstuff-dev:armhf : Depends: libkf5newstuff5:armhf (= 5.11.0+git20150702.1035+15.04-0) but it is not going to be installed
                            Depends: libkf5xmlgui-dev:armhf (>= 5.11.0~) but it is not going to be installed
 libkf5notifications-dev:armhf : Depends: libkf5config-dev:armhf (>= 5.11.0~) but it is not going to be installed
                                 Depends: libkf5coreaddons-dev:armhf (>= 5.11.0~) but it is not going to be installed
                                 Depends: libkf5notifications5:armhf (= 5.11.0+git20150702.1117+15.04-0) but it is not going to be installed
                                 Depends: libkf5windowsystem-dev:armhf (>= 5.11.0~) but it is not going to be installed
 plasma-framework-dev:armhf : Depends: libkf5package-dev:armhf (>= 5.11.0~) but it is not going to be installed
                              Depends: libkf5plasma5:armhf (= 5.11.0+git20150702.1347+15.04-0) but it is not going to be installed
                              Depends: libkf5plasmaquick5:armhf (= 5.11.0+git20150702.1347+15.04-0) but it is not going to be installed
                              Depends: libkf5service-dev:armhf (>= 5.11.0~) but it is not going to be installed
```

aleixpol, 2015-07-06:
Here's an easier way to test the sdk:
```
git clone git@github.com:plasma-mobile/xbuilder.git
cd image
docker build -t pms .
docker run -ti pms bash
```

HTH

apachelogger, 2015-07-07:
Oh oh @aleixpol btw, architecture labels on deb-src lines make no sense, sources have no architectures :stuck_out_tongue:

perhaps you should remove them, although I doubt they have a negative impact

closed 2015-07-07 by apachelogger