blue-systems/plamo-initial-dev_DEPRECATED #130: PolicyKit doesn't show an agent, we have a bad ConsoleKit

aleixpol, 2015-06-26:
I've been trying to figure out why PackageKit doesn't work, then realized it didn't work because of PolicyKit.

```
phablet@ubuntu-phablet:~$ pkexec --disable-internal-agent ls
Error executing command as another user: No authentication agent found.
```

Long story short, here's the issue and a link to the fix.
Issue: https://github.com/ConsoleKit2/ConsoleKit2/issues/26
Fix: https://github.com/EricKoegel/ConsoleKit2/commit/7c33506401269c36274987228097042fd23897c7

It could be interesting to package it and see if it fixes the issue.

apachelogger, 2015-06-29:
@aleixpol I think I'll need an adjusted patch for consolekit 0.4

 07:25:25  ...mobile/tmp/consolekit-0.4.6 
$ quilt push
Applying patch debian/patches/0001-CkSession-fix-uid-handling.patch
patching file src/ck-session.c
Hunk #1 FAILED at 55.
Hunk #2 FAILED at 386.
Hunk #3 FAILED at 1244.
Hunk #4 succeeded at 1302 with fuzz 1 (offset 19 lines).
3 out of 4 hunks FAILED -- rejects in file src/ck-session.c
Patch debian/patches/0001-CkSession-fix-uid-handling.patch does not apply (enforce with -f)

aleixpol, 2015-07-02:
Can we try with this one?
https://paste.kde.org/p83gteghp

It does the same as the patch in master (for 0.9) although the codebases have diverged quite a bit since...

apachelogger, 2015-07-06:
https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5187937/+listing-archive-extra

@aleixpol fails to build on all architectures :/

aleixpol, 2015-07-06:
Correct, here's a new attempt, this time I checked it builds locally: http://pastebin.com/AeCcizm0

apachelogger, 2015-07-07:
it failed again :/

https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5195431/+listing-archive-extra

diff between packages

https://launchpadlibrarian.net/210971138/consolekit_0.4.6-5ubuntu0%2Bkubuntu1_0.4.6-5ubuntu0%2Bkubuntu2.diff.gz

aleixpol, 2015-07-08:
I think this one will be for good: http://pastebin.com/Udphbapr

I didn't know I couldn't just add/remove/modify the patches and that I had to use `quilt` for it... 8-)

star-buck, 2015-07-08:
@aleixpol : with this fixed, will muon discover work?

apachelogger, 2015-07-08:
:sob: /«PKGBUILDDIR»/src/ck-session.c:824: undefined reference to `console_kit_session_get_unix_user'

https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+build/7631506

aleixpol, 2015-07-08:
@star-buck I can't assess how much it will work without experiencing it running, but at least we can get it running with the desktop version information, which is quite a leap already. I'm especially interested in seeing how good the click backend is so that we can un/install applications from there, although I fear the plugin they made is only half-baked.

@apachelogger I feel so clumsy...

aleixpol, 2015-07-08:
http://pastebin.com/qdpfYdzT this seems to work, consolekit seems to have diverged quite a bit since the version we're using. I'm just hoping we'll get to use systemd soon...
@apachelogger Thanks for bearing with me =.=

apachelogger, 2015-07-09:
:heart: https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5200152/+listing-archive-extra

apachelogger, 2015-07-10:
@aleixpol bad news. this might have broken something on amd64 where CK is now using 100% CPU http://i.imgur.com/zYIPNzt.jpg (haven't actually checked on the phone, so it might be the same there)

@mgraesslin also seems to have problems with CK on his exopc [11:02] <mgraesslin> and I had to kill CK because it was unusable slow with it

aleixpol, 2015-07-10:
@apachelogger *sadness* how about packaging the last version? Would it be hard? Otherwise I can spend some time next week trying to see what's different between us and Ubuntu Phone...

apachelogger, 2015-07-14:
let's see I guess

it being called CK2 doesn't really instill confidence in its smooth compatibility though

apachelogger, 2015-07-14:
literally none of the patches apply anymore and I have no clue if they are still relevant. it produces packages though locally. let's see what launchpad makes of it

https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5211180/+listing-archive-extra

apachelogger, 2015-07-14:
fixed build

https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5211185/+listing-archive-extra

aleixpol, 2015-07-14:
so if I dist-upgrade I'll get to test the new consolekit?

apachelogger, 2015-07-14:
@aleixpol that's the idea, yes

aleixpol, 2015-07-16:
I tested it, doesn't work either. :cry: 

Will keep trying...

apachelogger, 2015-07-17:
Leaving the CK2 build in for now. It doesn't seem to cause any problems anyway.

aleixpol, 2015-07-17:
It doesn't seem to work on Ubuntu Phone either...

apachelogger, 2015-07-17:
oh oh oh, could it be that something requires an upstart job that doesn't have one anymore? (i.e. only systemd unit where we'd need an upstart one)

bhush9, 2015-09-15:
So.. I debugged this and actually cause for this is quite stupid.. :laughing: 

On normal plasma desktop we ship this file

```
/etc/xdg/autostart/polkit-kde-authentication-agent-1.desktop
```

Which starts.

```
%libexecdir%/polkit-kde-authentication-agent-1
```

without it there would be no password dialog.. I've patch that starts it from plasma-phone.. Inserting mandatory screenshot.

[image]

Now thing is this doesn't get focus.. I will try to figure it out..

bhush9, 2015-09-15:
http://commits.kde.org/plasma-phone-components/d4d7c4e7bcef92ff91edb54fb4198c42805391a7

bhush9, 2015-09-15:
So actually it is focused but frozen

quick backtrace of polkit-kde-authentication-agent-1 shows

```
#0  0xb607a4e2 in poll () from /lib/arm-linux-gnueabihf/libc.so.6            
#1  0xb345f7b6 in wl_display_dispatch_queue ()                               
   from /usr/lib/arm-linux-gnueabihf/libwayland-client.so.0                  
#2  0xb34e7e1c in QWaylandDisplay::blockingReadEvents() ()                   
   from /usr/lib/arm-linux-gnueabihf/libQt5WaylandClient.so.5                
#3  0xb34e9ef2 in QWaylandWindow::waitForFrameSync() ()                      
   from /usr/lib/arm-linux-gnueabihf/libQt5WaylandClient.so.5                
#4  0xb34e290e in QWaylandShmBackingStore::beginPaint(QRegion const&) ()     
   from /usr/lib/arm-linux-gnueabihf/libQt5WaylandClient.so.5                
#5  0xb696d436 in ?? () from /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5 
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

According to MartinG : it's frozen waiting for kwin sending a frameRendered which kwin will never do and perhaps can be solved with Qt 5.5

mgraesslin, 2015-09-16:
I just tried to reproduce on the exopc (Kubuntu 15.04) and are not able to reproduce it there. As it's Qt 5.4 my assumption that Qt 5.5 might fix it, clearly doesn't hold.

bhush9, 2015-09-18:
Okay so...

I installed latest image and it seems to work pretty well. However @mgraesslin says it crashes completely for him sometimes which I can't reproduce at all. I am not sure what to do. I think this needs more testing.

mgraesslin, 2015-09-18:
after fixing the broken update I don't hit the problem mentioned in previous comment any more.

star-buck, 2015-11-26:
then closing this.

closed 2015-11-26 by star-buck