blue-systems/plamo-initial-dev_DEPRECATED #131: [malitt]: does not come up when entering wifi-password in top-panel

star-buck, 2015-06-26:


star-buck, 2015-07-07:
also doesnt come up when entering "search" field

mgraesslin, 2015-07-08:
yes, those two are clearly related.

notmart, 2015-07-08:
the dialer app has a plasma component text field and there the keyboard pops up fine.
@mgraesslin is it possible that plasmashell itself being launched by kwin has the environment cleared somehow and doesn't have the proper input method set?

notmart, 2015-07-08:
it seems to have the proper qt_im_module

notmart, 2015-07-08:
i can confirm minputcontext gets actually loaded, and sometimes i have the keyboard correctly popping up for the search field, seems to break after some times.
on the wifi password the times it does pop up, the text is not entered inthe field

mgraesslin, 2015-07-09:
@notmart the problem is that KWin does not pass keyboard focus to "Desktop" and "Dock". The reason for this change was that we don't want to get them closed. So I think we need to remove the restriction on keyboard focus again, but at the same time find a better way to mark the window as not closeable. Let's discuss once you're on IRC.

luebking, 2015-07-09:
>
> keyboards are by their nature transient utility windows (on top of their
> master and hidden for sure when their group is inactive) - i also do not
> see, why they should not be closable (to eg gain access to what's behind).
> it's only important that activating (clicking) the (even focused) input
> element re-shows them.
>

> if one wants a "global" keyboard (eg. inside a panel) it imo needs to work
w/o being active or focused - to maintain the concept where the input will
go

mgraesslin, 2015-07-09:
Actually what I wrote above is completely wrong. We check for closeable already on dock/desktop. And there seems to be no restriction on desktop/panel not getting focus.

@luebking The problem is that Qt's input method integration only requests the virtual keyboard to show if the widget has the active keyboard focus. And on the QtWayland plugin this requires the wl_seat's keyboard focus which is bound in kwin to the Workspace::activeClient(). (I find it very questionable that the virtual keyboard only shows on keyboard focus, but well...)

mgraesslin, 2015-07-09:
I have just triggered another kwin build which should improve the situation.

On the desktop I get in a reliable way the virtual keyboard to show on the search field on the second (!) touch event on the search field.

On the top panel I still do not get the virtual keyboard to show. It looks like the input field doesn't react on the touch event (also not to mouse events), thus doesn't get focus.

notmart, 2015-07-14:
latest kwin still doesn't give keyboard focus to the panel window, the text field doesn't get focus at all

mgraesslin, 2015-07-14:
I doubt it's a kwin issue. There is nothing which would prevent it. Feel free to check kwin code yourself (four eyes see more than two).

For me it's clearly the input fields not reacting to the touch events (I'm testing with networkmanager applet). Which would rather indicate a problem in QtQuick and friends.

luebking, 2015-07-14:
For the records: if that window is dock type, Workspace::takeActivity will strip ActivityFocus (unless there's ActivityFocusForce) and *not* invoke *Client::takeFocus()

KWin doesn't activate/focus docks (easily)

mgraesslin, 2015-07-15:
Thanks Thomas, that might in deed be the problem here.

So the pull down panel needs either to become force activated (how?) or become not a panel (maybe it could be fullscreen window instead?)

notmart, 2015-07-15:
changed the top sliding panel window type, the keyboard works again.

closed 2015-07-15 by notmart

luebking, 2015-07-15:
If you can intercept the activation demand (eg. when showing?), KWindowSystem::forceActiveWindow() should do - I think that's also wired to some QML component somewhere.

mgraesslin, 2015-07-16:
> KWindowSystem::forceActiveWindow() 

not yet implemented on Wayland :-(