blue-systems/plamo-initial-dev_DEPRECATED #132: [Settings]: Screen: doesn't work, exiting kcm crashes Settings

star-buck, 2015-06-26:


sebasje, 2015-07-07:
Can't reproduce it here, any special steps you've taken?

star-buck, 2015-07-07:
you start the Settings app, you can enter any kcm and press the "X" close button just fine.
You enter the Screen kcm, then hit the close button and after 2-3 seconds of closing, a crash notice appears.

sebasje, 2015-07-07:
Got it, I can reproduce.

star-buck, 2015-11-15:
img11: still crashes the same

sebasje, 2015-11-15:
The bug should be fixed now, but it probably hasn't made it into the image yet.

bhush9, 2015-11-15:
@sebasje : any specific commit?

mgraesslin, 2015-11-16:
http://commits.kde.org/kscreen/4653c287f844f2cb19379ff001ca76d7d9e3a2a1

mgraesslin, 2015-11-16:
which might not be the same, I misinterpreted this to be the kscreen kcm in systemsettings

sebasje, 2015-11-16:
Yes, that's the commit I was thinking of. I had debugged this problem at some point, but couldn't figure out the cause, if I recall correctly, this seems to be the fix we needed. Let's try once package updates are available.

star-buck, 2016-03-02:
fixed.

closed 2016-03-02 by star-buck