blue-systems/plamo-initial-dev_DEPRECATED #133: [packages]: add intel arch on mobile.kci.pangea.pub

star-buck, 2015-06-27:
when we have intel packages, we can do an ISO for intel based tablet version of plasma-mobile (like former kubuntu plasma-active).

mgraesslin, 2015-06-28:
I wanted to try it on the wetab/exopc. It would be awesome to have that for Akademy as lots of KDE devs still have such devices.

There is just one "problem". We need a gles build of Qt as the intel driver on those devices only reports OpenGL 1.4 (but does OpenGL ES 2.0). This causes problems for example in KWin (doesn't enable compositing). That's the reason why I haven't installed it on the wetab yet as I hadn't found the time yet to recompile Qt.

star-buck, 2015-06-29:
@apachelogger :
do we need anything on (DO) serverside to offer intel based packages for mobile?

apachelogger, 2015-06-29:
@star-buck not sure. when I tried it on friday your DO account didn't quite work (jenkins refused to provision) but my private credentials worked sort of as a temporary droplet was created and able to run stuff (http://mobile.kci.pangea.pub:8080/job/dotest/1/console). For all I know your credentials not working might be a bug in the jenkins plugin though. In general the DO jenkins plugin isn't too exciting and has various problems that will need figuring out. There are also some open workflow questions Rohan and I have yet to resolve.

If we want any chance of getting this done before akademy I'd suggest going with 4 additional static 14.04 x32 droplets/servers instead of provisioning temporary droplets. We might also need x64 ones, I am not sure as I do not know how our repo software handles architecture:all packages.
Etherway, we will definitely need x32 droplets for now as I just remembered that the wetab is in fact i386 and not amd64 which makes everything so much more annoying :sweat_drops:

star-buck, 2015-06-29:
How about ms surface tablets, so we wont need 32bit for wetab right now if
that makes live easier?

apachelogger, 2015-06-29:
Seems they are 64bit so that is easier. Actually we might only need x64 droplets anyway. A quick web search suggests that we should be able to do 32bit builds via docker on 64bit though. So I think getting 4 x64 droplets is the way to go. I'd first attempt to get amd64 builds going anyway as we need it to resolve https://github.com/plasma-mobile/mobster/issues/128 as well.

star-buck, 2015-06-29:
So 4x64bit 14.04?
What Cpu and mem?

apachelogger, 2015-06-29:
4 times 2 GiB RAM, 2 CPUs

since we have server setup fully automated we can easily add or switch to different servers anyway, so the specs do not matter all that much for now.

star-buck, 2015-06-29:
here they are:

do-alpha-intel  46.101.251.195 
do-beta-intel  46.101.251.90 
do-gamma-intel  46.101.251.91 
do-delta-intel  46.101.251.92

star-buck, 2015-06-30:
@apachelogger :
how is the status to move forward?

davidedmundson, 2015-07-18:
I've pushed something into simplelogin that sets the VT to graphical that makes kwin_wayland on DRM work.

This is in master of simplelogin

I've tested this doesn't break the phone, so can be just updated everywhere.

apachelogger, 2015-07-20:
seems work thanks!

star-buck, 2015-08-13:
done.

closed 2015-08-13 by star-buck