blue-systems/plamo-initial-dev_DEPRECATED #137: [IMG]: 123 doesnt seem to boot?

star-buck, 2015-06-30:
after flashing, Google logo displays but nothing else

star-buck, 2015-06-30:
maybe bluez causes this?

jriddell, 2015-07-01:
could be bluez, building an image without bluedevil to see if that helps

previous images 122 was saved into kubuntu-phone/devel so that can still be booted

jriddell, 2015-07-01:
Doesn't seem to be anything to do with bluez

kwinwrapper starts plasma fine when you run it on the command line

running
simplelogin --exec kwinwrapper --user phablet
gives
starting  "kwinwrapper"
"initgroups( phablet ,  32011 ) failed for user:  "phablet""

jriddell, 2015-07-01:
Jan  3 04:08:45 ubuntu-phablet simplelogin: pam_unix(sddm-autologin:session): session opened for user phablet by (uid=32011)
Jan  3 04:08:45 ubuntu-phablet dbus[883]: [system] Rejected send message, 2 matched rules; type="method_call", sender=":1.41" (uid=32011 pid=2449 comm="simplelogin --exec kwinwrapper --user phablet ") interface="org.freedesktop.login1.Manager" member="CreateSession" error name="(unset)" requested_reply="0" destination="org.freedesktop.login1" (uid=0 pid=1183 comm="/lib/systemd/systemd-logind ")
Jan  3 04:08:45 ubuntu-phablet simplelogin: pam_systemd(sddm-autologin:session): Failed to create session: Access denied
Jan  3 04:11:33 ubuntu-phablet sudo: pam_extrausers(sudo:auth): authentication failure; logname= uid=32011 euid=0 tty=/dev/pts/11 ruser=phablet rhost=  user=phablet

from /var/log/auth.log

star-buck, 2015-07-01:
Any idea what might have changed from 122 to 123 to be causing this?

jriddell, 2015-07-01:
harald worked it out, a broken $PATH in /etc/environment, spinning a fix

jriddell, 2015-07-01:
image 126 all good, sorry for the hassle

closed 2015-07-01 by jriddell