blue-systems/plamo-initial-dev_DEPRECATED #138: [apps]: video player

star-buck, 2015-06-30:
jolla llsvplayer packages deb armhf:
https://app.box.com/s/uyy8vrg87jxh8r04jjfqgpq7m0g60c8d
https://app.box.com/s/o71qozqaxblq8ikm76c6i88k35klc7q0

@jriddell : 
please include packages in new IMG

jriddell, 2015-07-02:
jolla app
https://github.com/llelectronics/videoPlayer
needs libsailfishapp
https://github.com/sailfish-sdk/libsailfishapp
needs mlite5
https://github.com/nemomobile/mlite
and maybe more dependencies

jriddell, 2015-07-02:
the .debs install fine but I can't get it to do anything

phablet@ubuntu-phablet:~$ vplayer  --help
Created Inout context: QComposeInputContext(0x16e8010)

it then quits

star-buck, 2015-07-16:
@llelectronics :
Can you help with how to make your packages work?

llelectronics, 2015-07-16:
@jriddell please see the https://github.com/llelectronics/videoPlayer/tree/plasma (Plasma Tree here for the source code)

As for the binary. 
./vplayer --phone should start the ui (when on desktop. On the phone the PLASMA_PLATFORM variable should be set so it should run without a problem)

star-buck, 2015-07-21:
@jriddell l:
is the videoplayer shipped?

star-buck, 2015-07-30:
@apachelogger :
Lets build packages of videoplayer in Mobile CI:
https://github.com/llelectronics/videoPlayer/tree/plasma

star-buck, 2015-07-31:
@apachelogger :
Please build videoplayer from here:
https://github.com/plasma-apps/videoPlayer/tree/plasma

so package can be found in Mobile CI repo here:
http://mobile.kci.pangea.pub/pool/main/

llelectronics, 2015-07-31:
I already added https://github.com/plasma-phone-packaging/videoPlayer-packaging 
So it just needs to be added to jenkins I guess :)

star-buck, 2016-03-02:
@bhush9 : please try to build and ship this so we have a video player to test things on

bhush9, 2016-03-04:
I've shipped it in IMG28 but doesn't start.

llelectronics, 2016-03-04:
Any error message in the terminal?

bhush9, 2016-08-01:
I've fixed it at packaging level to pass --phone to command line... it somehow fails to start even if PLASMA_PLATFORM=phone is set.

bhush9, 2016-08-01:
In next image it will be working.

star-buck, 2017-01-10:
i still dont see it shipped, please ship again.

closed 2017-12-09 by star-buck