blue-systems/plamo-initial-dev_DEPRECATED #141: kwinwrapper hardcodes arm

apachelogger, 2015-07-07:
kwinwrapper presently hardcodes armhf paths which is meh on amd64 builds:

> export  QML2_IMPORT_PATH=/usr/lib/arm-linux-gnueabihf/qt5/qml/:$QML2_IMPORT_PATH

probably kwinwrapper needs to be changed to a cmake configure_file file and then use dpkg-architecture to get the target architecture at build time. or instead of usign dpkg-architecture expect the callee (debian/rules) to tell it which path to use.

bhush9, 2015-10-01:
What I see is this line is utterly useless.. there is no need to do any dpkg/cmake magic with it IMO.

Thing is QML2_IMPORT_PATH it specifies is already in qt's own qml import lookup path. So its just useless..

I just tested by removing this line locally and seems to work fine. I will remove this from upstream repo and close this issue when is packaged.

bhush9, 2016-01-20:
Closing this now. I've removed that line long ago.

closed 2016-01-20 by bhush9