blue-systems/plamo-initial-dev_DEPRECATED #144: [networking] prefer wifi over GSM for data

sebasje, 2015-07-07:
When I'm connected to a broadband network and a wifi network, the mobile connection takes precedence in the routing table. We should make sure that wifi is preferred, as wifi is usually lower cost and higher bandwidth.

bhush9, 2015-08-03:
Possible fix.

in, ``` /etc/NetworkManager/conf.d/51-prefer-wifi.conf ``` file,

```
[connection-wifi]
match-device=type:wifi
ipv4.route-metric=25
ipv6.route-metric=25

[connection-gsm]
match-device=type:gsm
ipv4.route-metric=50
ipv6.route-metric=50
```

bhush9, 2015-08-03:
(Note: I didn't test this, please test first)

sebasje, 2015-08-05:
Will test and report back here.

bhush9, 2015-09-24:
IIRC @sebasje reported this was auto magically fixed for him on IRC.. can you confirm? Otherwise I will ship this configuration in next image.

star-buck, 2016-01-10:
is it shipped? then please mark 'Ready'.

bhush9, 2016-01-10:
I think this issue is fixed, and I haven't seen this recently.. so shipping that config doesn't seem to be required.

closed 2016-01-10 by star-buck