blue-systems/plamo-initial-dev_DEPRECATED #146: [packaging] Please package new repository kwayland-integration

mgraesslin, 2015-07-09:
Hi,

I just created a new repository under kde/workspace called kwayland-integration. This should be added to packaging and installed by default in the mobile image.

The dependencies are:
* kwindowsystem
* kidletime
* kwayland

It is required for providing support for idle time and a proper kwindowsystem integration plugin.

apachelogger, 2015-07-14:
new package available. also kidletime and kwindowsystem recommend kwayland-integration now, as discussed on IRC.

due to recommends this should automatically end up on the image, assigning to @jriddell to make sure it does.

jriddell, 2015-07-14:
hmm doesn't seem to be installed on new image I made today although libkf5idletime5 is and that recommends kwayland-integration

jriddell, 2015-07-15:
<Riddell> uh something tells apt not to install recommends
<Riddell> phablet@ubuntu-phablet:/etc/apt/apt.conf.d$ cat 00recommends 
<Riddell> APT::Install-Recommends "false";

rebuilding with that file removed

apachelogger, 2015-07-15:
[13:26] <Riddell> I wonder what that is
[13:26] <sitter> probably to reduce image size
[13:26] <sitter> not entirely unreasonable
[13:27] <sitter> possibly
[13:27] <sitter> simplelogin sould simply depend on kwayland-integration

jriddell, 2015-07-16:
on current image

closed 2015-07-16 by jriddell