blue-systems/plamo-initial-dev_DEPRECATED #148: [packaging] plasma-settings package is empty

sebasje, 2015-07-09:
The plasma-settings package is empty since the last change (Harald enabled debug symbols, I suppose something went wrong with that). The files are neither in the package, nor in its dbg package.

-rw-r--r-- 1 root root   83840 Jul  8 03:34 plasma-settings_0.0.1ubuntu4+git20150708.0115+15.04-0_armhf.deb
-rw-r--r-- 1 root root    2102 Jul  9 13:13 plasma-settings_0.0.1ubuntu4+git20150709.1055+15.04-0_armhf.deb

Reinstalling this previous version fixes it (no dbg though):
plasma-settings_0.0.1ubuntu4+git20150709.1055+15.04-0_armhf.deb

sebasje, 2015-07-09:
build error log:
http://mobile.kci.pangea.pub:8080/job/vivid_unstable_plasma-settings-packaging_pub/lastFailedBuild/console

apachelogger, 2015-07-10:
http://mobile.kci.pangea.pub:8080/job/vivid_unstable_plasma-settings-packaging_pub/12/

closed 2015-07-10 by apachelogger