blue-systems/plamo-initial-dev_DEPRECATED #150: New dependencies for plasma-settings

mck182, 2015-07-13:
Now it requires qml-module-org-kde-kaccounts and signon-ui and either packaged kaccounts-providers OR make account-plugin-google a dependency. kaccounts-providers would be preferred.

apachelogger, 2015-07-13:
I am too stupid to close bug reports. 

ANYWAY
so, since I wasn't sure wether we need the kded and since it is altogether more sensible we are depping on kaccounts-integration (kded&kcm) as it also already has a complete list of all dependencies that we supposedly need.

@mck182 it would probably be worthwhile to look at the package lineup though and if the kded is needed for mobile we should probably restructure such that the kded brings in all the runtime deps?

closed 2015-07-13 by apachelogger

mck182, 2015-07-13:
That doesn't seem quite right, the kded module just reacts on when account is added/removed and triggers the sync jobs. There is no actual dependency of the kded on any of these runtime things. However there will be a plugin for the kded module and this plugin will be part of plasma-settings, so that plugin could technically have the runtime deps pulled in. Still doesn't seem right but if it helps packaging, I'll do it.

apachelogger, 2015-07-13:
Well, the package also contains the KCM which is why the deps are on kaccounts-integration right now since I suppose the KCM is the closest thing to a dependee we have.
That being said, either the deps move to the kded or we'd have to make a pseudo-package who's only purpose is to allow us to pull in kaccounts dependencies. Neither is more work, I do however hate clutter packages.

mck182, 2015-07-13:
Right. Ok, I'll look into it.