blue-systems/plamo-initial-dev_DEPRECATED #151: Keyboard does not come up in a webview

mck182, 2015-07-13:
KAccounts uses signon-ui to register a Google account. Signon-ui opens a webview for Google login form, but the keyboard does not come up when tapping the input fields, which prevents adding the Google account.

Running it from konsole there's nothing indicating a problem, the input field correctly receives focus and the blinky cursor appears, just no keyboard.

mgraesslin, 2015-07-13:
what is a "webview"?

mck182, 2015-07-13:
QWebView class/subclass.

http://doc.qt.io/qt-5/qwebview.html

mgraesslin, 2015-07-13:
I just tried the fancybrowser of the qtwebkit5-examples examples on the phone. Navigated to google.de, clicked the search field, keyboard opened and my input got accepted.

Can you please verify that the application you use has the QT_IM_MODULE environment variable set (maybe dbus activated and env variables missing?)

mck182, 2015-07-13:
It has QT_IM_MODULE=maliitphablet - while I can see that plasmashell has just "maliit"

It is DBus activated by signond iirc which also has "maliitphablet". So I assume that is wrong, right?

I think we could perhaps preload signond from active-settings, on the module opening.

mgraesslin, 2015-07-14:
yeah there is clearly something wrong here. The env variable is wrong. We need to figure out where it is coming from and fix it :-)

mgraesslin, 2015-07-14:
as an idea: can you modify the plasma-phone script to call ksyncdbusenv after setting the env variables. That should set our QT_IM_MODULE and overwrite whatever was set by ubuntu.

mck182, 2015-07-14:
> We need to figure out where it is coming from and fix it :-)

It's being set in /etc/environment

> call ksyncdbusenv after setting the env variables

That doesn't seem to do anything unfortunately.

---
Turns out simply preloading signon-ui with plasma won't work without some extensive patcheroo to signon-ui because a) it closes on timeout b) it closes when you log in (to google) c) it closes when the user closes the window.

So, the best course of action would now be to fix the root of the issue and that is changing it in /etc/environment or making the plugin actually maliitphablet, that would seem to require one less env var override from us.

jriddell, 2015-07-15:
added echo QT_IM_MODULE=maliit >> /etc/environment to mobster scripts, will be in next build

closed 2015-07-15 by jriddell