blue-systems/plamo-initial-dev_DEPRECATED #152: kci plasma-phone-meta

jriddell, 2015-07-14:
plasma-phone-meta is the lovely new meta package to replace the long list in chroot-remaster, I've uploaded it using on demand but please add it to mobile.kci
    "plasma-phone-meta":
        ["git@github.com:plasma-mobile/plasma-phone-meta.git",
        "git@github.com:plasma-phone-packaging/plasma-phone-meta.git",
        "kubuntu_unstable"]

jriddell, 2015-07-14:
needs germinate installed on whatever builds the source packages
http://mobile.kci.pangea.pub:8080/job/vivid_unstable_plasma-phone-meta_src/lastFailedBuild/console

apachelogger, 2015-07-15:
why are we using germinate?

jriddell, 2015-07-15:
cos that's how seeds get turned into meta packages?

apachelogger, 2015-07-15:
but why are we using a seed? this seems vastly overengineered for what we are doing with it

jriddell, 2015-07-15:
cos a seed file is tidier than a long list of dependencies or a bash script

the problem is I couldn't work out how to get germinate to use git, it'll try but it won't add the .git on the end of the url so I had to use bzr on launchpad

apachelogger, 2015-07-15:
...overnegineering

http://manpages.ubuntu.com/manpages/precise/man5/deb-substvars.5.html

jriddell, 2015-07-15:
like the Forth Bridge it'll be a world heritage site one day

apachelogger, 2015-07-16:
as of http://mobile.kci.pangea.pub:8080/job/mgmt_docker_vivid/24/ the tooling should install germinate to generate sources

closed 2015-07-16 by jriddell