blue-systems/plamo-initial-dev_DEPRECATED #153: [kaccounts] Package patched signon-ui

mck182, 2015-07-14:
We will need to package our own signon-ui with my patches [1] after all. The unity stuff inside signon-ui crashes when finishing the auth session and that makes it unable to add the account at all. [2] is in case you just need a diff.

[1] - https://code.launchpad.net/~mardy/signon-ui/lp1473422
[2] - http://bazaar.launchpad.net/~mardy/signon-ui/lp1473422/revision/139

apachelogger, 2015-07-15:
FTR when patched we probably need to apt mark hold the patched version as to prevent other ubuntu-image induced packages from overwriting our patched version until ubuntu adopts the presented patch.

apachelogger, 2015-07-16:
https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5216393/+listing-archive-extra

@jriddell please make sure to mark hold it as mentioned in the previous comment

jriddell, 2015-07-16:
signon-ui_0.17+15.04.20150410-0ubuntu1+phone1_armhf.changes uploaded to mobileci
mobster script now has:
apt-get install -y signon-ui-service=0.17+15.04.20150410-0ubuntu1+phone1 signon-ui-x11=0.17+15.04.20150410-0ubuntu1+phone1

mck182, 2015-07-21:
I just noticed that signon-ui failed to build on arm, so it is not on the phone --> https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+build/7658885

Not entirely sure why, the buildlog says

libEGL warning: DRI2: failed to create any config
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)

star-buck, 2015-07-31:
@jriddell :
Has this been solved?

mck182, 2015-07-31:
The latest signon-ui armhf build is the one linked above, the failed one, so it's not solved.

Alternatively, the patch is stuck upstream waiting for QA at https://requests.ci-train.ubuntu.com/static/dashboard.html#?q=ubuntu%2Flanding-031

I don't know who to ping to get it moving though.

star-buck, 2015-09-19:
is this still pending?

mck182, 2015-09-23:
Yes, signon-ui is still failed at https://launchpad.net/~plasma-phone/+archive/ubuntu/ppa/+sourcepub/5216393/+listing-archive-extra

I'm not aware if it's being build elsewhere.

Also, signon-ui for wily already has that patch merged in, but not sure if that helps us anyhow as we're not switching to wily anytime soon.

jriddell, 2015-09-24:
silly qemu breaking in launchpad.  building now in on-demand

jriddell, 2015-09-24:
added to seed, plasma-phone-meta updating now

mck182, 2015-09-25:
Are you sure the package has the patch actually applied? signon-ui works as if the patch wasn't built with it. Also building signon-ui from master (which is the package plus just with the patch in question) makes it work correctly as expected. So I have a reason to believe the patch is not there.

Also, can debug package be added please?

mck182, 2015-09-28:
The signon-ui 0.17+15.04.20150410-0ubuntu1+phone2 still does not have the required patch; it still doesn't work.

Also now this version is not installable without specifying the full version. apt-cache follows:
```
phablet@ubuntu-phablet:~$ apt-cache policy signon-ui
signon-ui:
  Installed: 0.17+15.04.20150410-0ubuntu1
  Candidate: 0.17+15.04.20150410-0ubuntu1
  Version table:
     0.17+15.04.20150410-0ubuntu1+phone2 0
        500 http://mobile.kci.pangea.pub/ vivid/main armhf Packages
        100 /var/lib/dpkg/status
     0.17+15.04.20150410-0ubuntu1+phone1 0
        500 http://ppa.launchpad.net/plasma-phone/ppa/ubuntu/ vivid/main armhf Packages
 *** 0.17+15.04.20150410-0ubuntu1 0
        500 http://ports.ubuntu.com/ubuntu-ports/ vivid/main armhf Packages
       1001 http://ppa.launchpad.net/ci-train-ppa-service/stable-phone-overlay/ubuntu/ vivid/main armhf Packages
```

jriddell, 2015-09-29:
pinning rules broken :(  fixing now..

Is this the patch you need? it is in the package 0.17+15.04.20150410-0ubuntu1+phone2
http://embra.edinburghlinux.co.uk/~jr/tmp/lp1473422.patch

star-buck, 2015-11-15:
@jriddell : is this solved?

mck182, 2016-01-14:
This appears to be solved in latest image/package update.

closed 2016-01-14 by mck182